the-modem-distro/pinephone_modem_sdk #114: make aboot can't find libgcc.a on 0.6.6

jg110, 2022-06-16:
Trying `make everything` or `make aboot` gives the following error with 0.6.6 checked out:

```
NOTE: Executing Tasks
ERROR: aboot-1.0+git-AUTOINC+a4aeb00dfe-r0 do_compile: oe_runmake failed
ERROR: aboot-1.0+git-AUTOINC+a4aeb00dfe-r0 do_compile: ExecutionError('/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093', 1, None, None)
ERROR: Logfile of failure stored in: /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/log.do_compile.595093
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 12 mdm9607 -C /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/git LIBGCC=/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/recipe-sysroot/usr/lib/arm-poky-linux-gnueabi/11.2.0/libgcc.a ENABLE_HARD_FPU=1 TOOLCHAIN_PREFIX=arm-poky-linux-gnueabi- SIGNED_KERNEL=0 DEBUG=1 ENABLE_DISPLAY=0 WITH_DEBUG_UART=1 BOARD=9607 SMD_SUPPORT=1 MMC_SDHCI_SUPPORT=0
| make: Entering directory '/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/git'
| make[1]: Entering directory '/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/git'
| including app/aboot dev/keys dev/pmic/pm8x41 dev/vib lib/debug lib/heap lib/libc lib/libfdt lib/libufdt lib/openssl lib/ptable
| including lib/openssl/crypto lib/zlib_inflate
| generating build-mdm9607/config.h
| generating build-mdm9607/system-onesegment.ld
| linking build-mdm9607/lk
| arm-poky-linux-gnueabi-ld.bfd: cannot find /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/recipe-sysroot/usr/lib/arm-poky-linux-gnueabi/11.2.0/libgcc.a: No such file or directory
| make[1]: *** [make/build.mk:16: build-mdm9607/lk] Error 1
| make[1]: Leaving directory '/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/git'
| make: *** [makefile:20: make-make] Error 2
| make: Leaving directory '/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/git'
| ERROR: oe_runmake failed
| WARNING: /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093:180 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
|       #1: bbfatal_log, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093, line 180
|       #2: die, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093, line 165
|       #3: oe_runmake, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093, line 170
|       #4: base_do_compile, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093, line 148
|       #5: do_compile, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093, line 142
|       #6: main, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/aboot/1.0+git-AUTOINC+a4aeb00dfe-r0/temp/run.do_compile.595093, line 190
ERROR: Task (/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/meta-qcom/recipes-bootloader/aboot/aboot.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 522 tasks of which 521 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/meta-qcom/recipes-bootloader/aboot/aboot.bb:do_compile
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code
```

The issue seems to be that there is no 11.2.0 gcc directory, but there is an 11.3.0 one with libgcc.a inside. The recipes all seem to call for 11.2.0 (I can't find the string "11.3.0" in any files) so I'm not sure why 11.3.0 would have been used.

Biktorgj, 2022-06-16:
Yocto 4.0 started with GCC 11.2.0 but has later on moved to 11.3.0
I had that patched locally but hadn't yet pushed it to the repo. If you resync the repo it should work now...

Sorry!

jg110, 2022-06-18:
This fixes the issue thanks!

closed 2022-06-18 by jg110