the-modem-distro/pinephone_modem_sdk #115: Feature: Silent SMS warning

tandy-1000, 2022-06-16:
It would be useful to have the ability to have the modem report silent SMS messages, also known as stealth pings.

Biktorgj, 2022-06-17:
That... depends.

If it's *flash-sms*, then you could enable sms logging via chat, and it would keep those self-disappearing messages in openqti.log
(persistent logging + sms logging would leave all your received + sent messages in /persist/openqti.log.* so you can retrieve them later).

It it's a class 0 silent SMS destined for the SIM, then no, as that doesn't get out of the baseband (goes baseband --> SIM directly), so not exposed to the userspace where I could get it

tandy-1000, 2022-06-17:
Yes that would be class 0 silent SMS messages.

Damn, that's a shame, do you think it would ever be possible to log these messages?

goapunk, 2022-06-23:
@Biktorgj I'm not very familiar with the quectel devices but since they use qualcomm chips (?) do they also come with qualcomms debug interface (/dev/diag) (at least quectel is mentioned here: https://osmocom.org/projects/quectel-modems/wiki/Diag )? If yes we could port SnoopSnitch to the pinephone.

Biktorgj, 2022-06-24:
I didn't know you could dump GSM frames directly from the diag interface!

We do have the diag interface and it's sitting there, doing nothing. It's not exposed to the outside world, but we certainly have complete access to it, so now I'm going to spend a week reading into how it wors :)

goapunk, 2022-06-24:
Exciting ;-) So we can add some mobile self-defense to the pinephone!