the-modem-distro/pinephone_modem_sdk #116: Bug: SMS received multiple times

tandy-1000, 2022-06-16:
I am receiving some SMS messages multiple times, also with a delay #80.

Let me know how to provide logs that are useful.

Biktorgj, 2022-06-17:
First I'd ask you to update to 0.6.7 (there were a few nasty bugs there), and make sure you have udev rules setup to avoid autosuspend (https://github.com/Biktorgj/pinephone_modem_sdk/blob/kirkstone/docs/SETTINGS.md)

Which phone are you running? and distro?

tandy-1000, 2022-06-17:
I am running the latest pmOS on a PP Manjaro Convergence Edition, I will upgrade and report back.
I did not set any udev rules, I will take a look at those.

BillFleming, 2022-10-16:
I recently saw this also on 0.68 on PPP on Danctnix arch, and I did install the udev rules. Bout to test 0.7 release on PP on danctnix