the-modem-distro/pinephone_modem_sdk #117: Can't hang up calls in 0.6.6

airtower-luna, 2022-06-16:
When I hang up a call on my Pinephone 1.2 with modem firmware 0.6.6 the call isn't terminated correctly:

* If a call hasn't been picked up yet, the phone keeps ringing on the remote side. If voicemail answers I get silence on my voicemail until its time limit.

* I let one call go to my voicemail and spoke for a few seconds, then hung up. The resulting voicemail had a few seconds of me talking, then silence to the time limit of the voicemail. So audio works both ways, call termination is the problem.

I see the same behavior with ADSP firmware 01.002.01.002, 01.003.01.003 (which I've been using successfully with 0.6.4), 30.004.30.004 and 30.006.30.006. I'm on the network of a O2/1&1 reseller in Germany, if it matters.

The attached logs are from the try with ADSP firmware 01.002.01.002.

* [modem-dmesg.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8919098/modem-dmesg.log)
* [openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8919099/openqti.log)
* [phone.daemon.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8919100/phone.daemon.log) (this is from pmOS syslog, I removed most output from services other than ModemManager, but left related NetworkManager lines)
* [phone-dmesg.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8919101/phone-dmesg.log)

I notice MM reports an "unexpected outgoing call".

Biktorgj, 2022-06-17:
Can you check if it still happens with 0.6.7?

airtower-luna, 2022-06-17:
It works fine again in 0.6.7, thank you very much! :smile_cat:

closed 2022-06-17 by airtower-luna

Grzechu3000, 2022-10-07:
55500

Grzechu3000, 2022-10-07:
60005

Grzechu3000, 2022-10-07:
No

Biktorgj, 2022-10-08:
Now we have to deal with spam here?