the-modem-distro/pinephone_modem_sdk #120: 0.6.7: can't sms to internal phone number

bircoph, 2022-06-19:
Hi!

After 0.6.5 -> 0.6.7 update I can no longer sms to internal number. I use new +223344556677 and sent AT+SIMUSMS after update (though I'm not sure it is necessary to send it after firmware update if it was activated before).

I use pinephone, pmos-22.06, phosh-0.17.0-r2.

See attached openqti.log: it is cut from the moment when "uptime" sms was sent to the phone and filtered from "Thermal zones" messages.
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8936320/openqti.log)

From what I see in the log, the reply is ready, but can't be sent and is killed by timeout. I have no idea why it is not sent, however. I updated from pmos-21.12-latest-sp to 22.06 together with firmware update. Maybe modem manager or phosh got too smart and rejects sms from invalid number?

bircoph, 2022-06-19:
I tested with 0.6.5: it doesn't work as well. So either modemmanager or phosh (or whatever else) refuses to get internal phone sms message. Meanwhile ordinary network sms works fine with both 0.6.5 and 0.6.7, with both sending and receiving.

bircoph, 2022-06-25:
Thanks, `AT+WIPESMS` helped.

It is strange though that modemmanager reported empty list _before_ wiping with AT command:
```
$ mmcli -m any --messaging-list-sms
No sms messages were found
```

closed 2022-06-25 by bircoph

Biktorgj, 2022-06-25:
Not sure about it, but it could be that the modem had a _partial_ message in receiving state (a piece of a multipart message), and modemmanager didn't show it? 

You can list messages with AT+CMGL=4 (use minicom/microcom to get the full response) if it happens again