the-modem-distro/pinephone_modem_sdk #121: 4G: voice calls and ussd doesn't work

bircoph, 2022-06-20:
Hi!

I tested the following ADSP firmware with my provider: 01.002, 01.003, 30.004, 30.006.

On all of the I have following results:
2G, 3G: sms (in+out) works, ipv4/ipv6 data works, voice calls (in+out) work, ussd works
4G: sms (in+out) works, ipv4/ipv6 data works; doesn't work: voice calls (in+out) work, ussd requiests

For USSD requests I just never got any reply (and likely no data is sent to baseband).
For 4G failed outgoing call I have immediate reject, for 4G incoming call (from another phone) I get either busy signal or no signal at all.

Any idea how to fix at calls or ussd on 4G?

This is openqti.log for a failed 4G call:
```
[901.0361] I call_service_hindler: Outgoing call to *890
[901.0449] I call_service_handler: Call #0: *890
[901.0454] W handle_call_pkt: Call 1 of 1: Outgoing
[901.0458] E handle_call_pkt: --> Unknown call type?+?"?
[901.0462] I handle_call_pkt: --> Attempt...
[901.0468] I use_external_codec: RT5616 codec not detected
[901.0547] E start_audio: Can't set mixer{. unknown call type 0
[901.0563] I call_service_handler: Outgoing call to [none]
```

This is log for successful 3G call: 
[3g-openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8941440/3g-openqti.log)

For ussd requests I have no entries in the logs.

sado1, 2022-06-22:
I can confirm that I am unable to make calls, with 4G only enabled in Phosh settings. This is likely due to not implementing VoLTE feature yet (hardware supports it).

This may become pretty important in the future - for example, in my country, one of the carriers will disable 3G completely at the end of 2023. (others don't do it yet, though)

bircoph, 2022-06-25:
I tried to enable VoLTE manually as described in the pine64 wiki:
https://wiki.pine64.org/wiki/PinePhone#VoLTE
```
AT+QMBNCFG="select","ROW_Generic_3GPP"
AT+QCFG="ims",1
[reboot]
```
It doesn't work, the log is the same.

My operator is not available on the list, so I used a generic one. I found that necessary profile is available at https://github.com/JohnBel/QualcommMBNs/tree/master/extracted/mbn in the discussion of issue #40 : https://github.com/Biktorgj/pinephone_modem_sdk/issues/40#issuecomment-1046315286

Is there any instruction available of how to sign and load a selected profile?

Biktorgj, 2022-06-26:
VoLTE is implemented in the modem's baseband, which is not touched by the custom firmware, so if it works in stock, it'll work in here.

@bircoph's comment is spot on, if you can't make calls _at all_ while in 4G it is probably because the modem is unable to correctly register through IMS. As I don't know what any of you have tried or what are you running, the next thing would be to try the different firmware versions in case you're either running a totally outdated firmware, or one that doesn't work particularly well with your carrier. [Here](https://github.com/Biktorgj/quectel_eg25_recovery/branches) you can see the different firmware versions we've found. I'd say start with 002 if you have a really old version, and go up from there to .003, .004 and .006, which is the latest firmware we've found

There are some carriers where data doesn't work with a newer firmware, while for others things work perfectly fine with the latest one (and times where no matter what you do it won't work)

@bircoph, you *shouldn't* need to do anything with those commands, as distros will already set those up for you most of the times. You can disable IMS from the configuration settings in eg25-manager, and that will allow you to stay in 4G and let the modem automatically switch to 3G for calling, though it isn't a proper solution.

Unfortunately, so far I haven't managed to find a way to sideload any of those profiles and get them to work in the modem. From what I've read (there's not really much information about all that available) there are some bits that might mark the specific model they can run on, so, short of finding another model to retrieve those profiles from, you're probably better off going to Quectel's forums and asking them directly to provide you with a profile for your carrier (though I can't say if they'll do it, but there are some threads there where it seems they privately provided to some people)

Subsentient, 2022-08-15:
This firmware used to work for me on the original PinePhone just fine. I upgraded to a PinePhone Pro, flashed 0.6.8, calls immediately disconnect, doesn't ring either. Downgraded to 0.6.5 which is what my last PinePhone ran and tried all DSPs, no dice.
I find that weird since it's the same model of modem, and it works fine on Quectel's unholy firmware. I really want the power savings of this firmware. :^(

My carrier is an old Consumer Cellular account that uses T-Mobile towers. I don't think they changed anything. Weird.
Must use the generic LTE profile, but it *used* to work on the old PinePhones with this carrier.

Any ideas @Biktorgj ?

Biktorgj, 2022-08-31:
The firmware *does* work in a Pinephone Pro :)
Unless your PPP came with a really old ADSP firmware, there should be no reason for the firmware to fail, can I get some logs?

Biktorgj, 2022-08-31:
@bircoph Can you head over to issue #40?

bircoph, 2022-09-04:
Just for the record from #40: in order for 4G-only VoLTE calls to work, I had to enable "internet calls, technological mark" option in my provider's account. Then only USSD problem is left, but it is minor.

closed 2022-09-04 by bircoph