the-modem-distro/pinephone_modem_sdk #122: open firmware changed behavior?

reivanen, 2022-06-23:
I was trying to update the ADSP firmware per the instructions on the releases page, and it hung at "<waiting for devices>"

Then i tried updating the userspace using gnome-firmware, and it failed after rebooting the modem with error "Message recipient disconnected from message bus without replying"

Ready to give up i decided to try downloading the firmware package and ran the flashall script, and to my surprise it worked.

Investigating, i saw it sends "fastboot oem stay", so then i modified the guide on the releases page from

    Open a root terminal and run: echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2 to enter fastboot mode
    Run fastboot flash modem NON-HLOS.ubi && fastboot reboot

To

    Open a root terminal and run: echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2 && fastboot oem stay
    Run fastboot flash modem NON-HLOS.ubi && fastboot reboot

And it worked.

So there seems to be either bug in gnome-firmware and the documentation, or the behavior of the firmware has changed so the old documentation&implementation in gnome-firmware don't work any more. I was on 0.6.5 when doing all this.

Biktorgj, 2022-06-24:
Sending AT+QFASTBOOT should be enough to keep the bootloader in fastboot mode for as long as you want. If that doesn't work out of the box is because _something else_ is deciding the modem is not responding and it's attempting to restart it, probably. Also notice fwupd is very strict with timings, so if the modem doesn't get into fastboot mode as fast as fwupd expects, it might show an error, even if there wasn't really any, it was just slow to reboot.

If using fwupd from the command line, you're prompted to send a report of the error if things like these happen. If you get the chance, please send the report so we can take a look at what happened!

By the way, which distro are you running?

reivanen, 2022-06-24:
yes, every previous time it was enough, but now using the firmware versions mentioned it did not work. I was running manjaro phosh.

Biktorgj, 2022-06-25:
The `fastboot oem stay` is just a precaution in case you're going from a mix of an old version of the firmware with a newer bootloader or the opposite.

That said, it seems we have either a sync issue with the bootflag between the userspace and the bootloader *and* a timing issue with fwup where it waits too much before starting to flash. I will investigate it further and get back with the findings

Biktorgj, 2022-06-25:
Update: Yep, I wasn't respecting the FASTBOOT_MODE flag in lk2nd, so if fwupd took < 2 seconds to start flashing it would have stayed there, but if it takes more time the modem will just keep booting before it has a chance of flashing.
Fixed here: https://github.com/Biktorgj/lk2nd/commit/ff36032b8596cf51595fdfffa60e44159a453409
Will be included in next release... sorry for the mess up!

closed 2023-03-08 by reivanen