the-modem-distro/pinephone_modem_sdk #124: ppp - 0.6.7 fails to install using gnome-firmware

henrythemouse, 2022-06-26:
Hi,

Not sure this has anything to do with the 0.6.7 package, still I thought it might be worth reporting. 

The 0.6.7 release shows up in the gnome-firmware app. I've used this app before to install other versions of this firmware with success (0.6.6 being the most recent). This time there is something that caused it to fail. The system is an up-to-date mobian testing installation. The attached file includes the logs as well as a screenshot of the gnome-firmware error.

The flashall script worked fine.
Thanks again for the update.

[0.6.7-logs.zip](https://github.com/Biktorgj/pinephone_modem_sdk/files/8987377/0.6.7-logs.zip)

reivanen, 2022-06-27:
https://github.com/Biktorgj/pinephone_modem_sdk/issues/122

Biktorgj, 2022-07-01:
Yeah, I messed up a thing in fastboot and doesn't really _obey_ the command to stay there when launching it from QFASTBOOT. 
It's fixed in code and will be in for the next release, sorry for that. 
A workaround you can use is running `fastboot oem stay` in a root terminal prior to starting an update and leaving it there in the background while you start the update via gnome-firmware

henrythemouse, 2022-07-01:
great work, thanks.

closed 2022-07-01 by henrythemouse