the-modem-distro/pinephone_modem_sdk #126: 0.6.7 caused no modem start after wake from suspend

snek100, 2022-06-28:
after installing 0.6.7 to my pinephone via gnome firmware on postmarketos it works, however after waking from suspend the modem is greyed out and will not work until the phone is rebooted.

snek100, 2022-06-28:
Upon further investigation, I have found that this only occurs when "location services" is on.

snek100, 2022-06-28:
This is quite unfortunate since this service is important for the gnome maps app.

barbastrejo7, 2022-06-28:
Did you do those recommendations? https://github.com/Biktorgj/pinephone_modem_sdk/blob/kirkstone/docs/SETTINGS.md#pinephone

tootbrute, 2022-07-02:
I did the modification to /usr/lib/udev/rules.d/80-modem-eg25.rules but this still happens to me, too.

is there anyway to get the modem back without rebooting? is there a log we can look at to show what is happening?

postmarketOS
pinephone64

snek100, 2022-07-14:
It seems that location services was not the problem. I tried   https://github.com/Biktorgj/pinephone_modem_sdk/blob/kirkstone/docs/SETTINGS.md#pinephone which had no effect, however "sudo rc-service eg25-manager restart" always works.

snek100, 2022-07-14:
location services was causing the consistent suspend issue, this issue seems to be unrelated, but causes the same greyed out modem problem.

snek100, 2022-07-14:
I've now downgraded to 0.6.6 to test weather the problem persists.

snek100, 2022-07-18:
After using 0.6.6 for a few days, I can confirm that this issue is only present in 0.6.7.

xnopasaranx, 2022-07-22:
happens to me as well, have been trying with 0.6.7 and the latest 0.6.8, both regularly lose the modem in deep sleep. I have reverted back to 0.6.6 now as I need to be reachable. Will try to get some logs once the call I am expecting has come in (delivery company)

xnopasaranx, 2022-09-09:
Okay I think this is a configuration problem, at least for me it was. There has been a new udev rules script that has been rolled out with several OS images and it contains an error. @snek100 check /usr/lib/udev/rules.d/80-modem-eg25.rules and change the trigger DRIVER to DRIVERS. Both are valid, but only DRIVERS works correctly for me.

neovalis, 2022-09-14:
> Okay I think this is a configuration problem, at least for me it was. There has been a new udev rules script that has been rolled out with several OS images and it contains an error. @snek100 check [...]

Unfortunately this didn't work for me.  I tried with 0.6.7 and 0.6.8 with ADSP Version 01.003.01.003