the-modem-distro/pinephone_modem_sdk #127: Unable to call number with automated pickup

twiclo, 2022-06-28:
I'm trying to call my bank. They answer with an automated system where you press a number to get to the department you're after. I get the first second of audio then the call drops. If I call the local branch it rings to their building first then after a couple minutes the call is forwarded to the automated system. Once I hit the automated system the call drops. This is on the latest firmware on a pinephone running postmarketos

twiclo, 2022-06-28:
I just downgraded to 0.6.6 and am experiencing the same issue

Biktorgj, 2022-07-01:
That doesn't seem to have anything to do with the firmware, but just in case, can I take a look at the logs for that call (so we can know if the midem is receiving some kind of disconnect when forwarding or something like that)? 

https://github.com/Biktorgj/pinephone_modem_sdk/blob/kirkstone/docs/TROUBLESHOOTING.md#getting-logs-to-help-with-an-issue