the-modem-distro/pinephone_modem_sdk #128: Failed firmware upgrade

g3blv, 2022-06-29:
I tried updating the modem firmware on a PinePhone (the original version) with the Gnome Firmware app. The app closed after I had entered the password to initiate the update. Restarting the app the modem wasn't listed instead I think it said Google Android something. At this point I restarted the phone which was probably a mistake. Running the Firmware app after restart shows "Failed to load device list. No detected devices" and there is no devices listed. Running `lsusb` returns a list but I can't see "Quectel" in the list. Is there some way I can fix this?

Biktorgj, 2022-07-01:
Can you post a full log of the Phone's kernel log? 
The "Android" device you saw is the modem in fastboot mode, it's normal while it's doing an update. It disappearing and never coming back is not, unless something went whacky with Gnome Firmware and it were still flashing when you rebooted... The kernel log should tell us more

As root, please run  `dmesg > kernel.log` and post it here when you have some time

g3blv, 2022-07-02:
Please find `kernel.log` below. I ran the command right after a restart. Please let me know what else I can provide.

[kernel.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/9032688/kernel.log)

Biktorgj, 2022-07-02:
Good news is the modem is still working fine :)
```
[   34.859868] usb 2-1: new high-speed USB device number 2 using ehci-platform
[   36.484586] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   39.629110] usb 2-1: USB disconnect, device number 2
```
Now, since it failed once I think the best you can do is manually flash everything just in case.
Go to https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.6.7 and download latest firmware package.
Also download latest ADSP firmware: https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_30.006.30.006/update/NON-HLOS.ubi

Now, once you have everything downloaded and unpacked, do the following:
1. Turn the first killswitch (modem), off, and then on again.
2. In a root terminal, run `fastboot oem stay &` so It stays running in the background
3. In that same terminal, run `systemctl restart eg25-manager`. After a few seconds, you should see an [OKAY] from fastboot.
4. Now flash the ADSP firmware, run `fastboot flash modem NON-HLOS.ubi`
5. Once it's flashed, rerun the 0.6.7 flash script `flashall`
The modem should flash and reboot, and you should have a functional modem on the latest firmware

g3blv, 2022-07-02:
I'm running postmarketOS v22.06 and it seems like `systemctl` is not installed. See blow. Or am I doing something wrong?

```
pine64-pinephone:~$ sudo -i
[sudo] password for user: 
pine64-pinephone:~# fastboot oem stay &
pine64-pinephone:~# < waiting for any device >

pine64-pinephone:~# systemctl restart eg25-manager
-ash: systemctl: not found
pine64-pinephone:~# 
```
Do you know if `systemctl` is replaced by something else on postmarketOS / Alpine?

I've installed `android-tools`.

closed 2022-07-02 by g3blv

BillFleming, 2022-07-18:
Biktorgi, please post some variation of your comment about recovery in the flashing instructions, maybe a readme specific to flashing the ADSP would be best.
https://github.com/Biktorgj/pinephone_modem_sdk/issues/128#issuecomment-1172865793
On my PPP I had no issue using the fwupd cli to flash 0.6.7, but it hung up trying to flash the new ADSP afterwards.
My Jan 2022 PPP had an older version than .01.002.01.002 FYI so probably most users will need to upgrade at least once, and fwupd won't cover that part of it.
On arch it seems we need to stop/restart that eg25-manager service and maybe ModemManager service for the fastboot commands to work properly.
I'm going to guess that fwupd could update the ADSP but you would need some separate new entry developed as the default can't also upgrade ADSP since not everyone can use the same version....

airtower-luna, 2022-07-18:
@BillFleming, the bug you see with 0.6.7 is probably the one reported in #122 and #124, basically the modem doesn't wait in fastboot mode. The workaround described there should be enough if you just want to flash the ADSP.