the-modem-distro/pinephone_modem_sdk #129: Phone vibrates non-stop until reboot when receiving a call while in a call

neovalis, 2022-07-01:
0.6.5 PPP Mobian (This issue also existed in previous versions) If I'm in a call and a new call comes in the phone starts vibrating relentlessly without stopping.  For few seconds I can no longer hear the other person and they can't hear me, a few seconds later if I listen very closely through the loud vibration they can hear me I can hear them.  I get a text message from the modem saying "call from ***** while you're talking (ignoring it)".  I've always had to resort to rebooting the phone to stop the non-stop vibration when this happens.  I searched the open issues and didn't see this issue listed.

hooger, 2022-07-01:
I think this is an issue in the Calls/ModemManager stack, there are multiple issues for it (https://gitlab.gnome.org/GNOME/calls/-/issues/428, https://gitlab.gnome.org/GNOME/calls/-/issues/233).

However https://github.com/Biktorgj/pinephone_modem_sdk/releases 0.6.6 implemented a call rejection mechanism for the second call. I think Biktor will ask you to test, if your issues is present even with that fw version. If it disappears, than he did everything that can be done on the firmware side. At least that is my opinion. Hope it helps!

Biktorgj, 2022-07-01:
@neovalis Sending `callwait auto ignore` in the chat should make the modem send you a text message when you have a call waiting, but avoid passing that information to ModemManager, making the phone _not_ ring when you get that second call. Can you try that?

bircoph, 2022-12-27:
> @neovalis Sending `callwait auto ignore` in the chat should make the modem send you a text message when you have a call waiting, but avoid passing that information to ModemManager, making the phone _not_ ring when you get that second call. Can you try that?

I use phosh and have the same issue: it can't handle properly a second call while in active call (and vibrates phone until reboot which is really annoying). I tried `callwait auto reject` — it works for me: new calls are being automatically rejected.

I wonder what will happen with second call in `callwait auto ignore` mode? Will it be passed to the phone when first call will end or will it hang untill a caller or a network will cancel the call?

In reject mode I had an occasional problem twice: it looks like some callers use automated redial algorithm, so the modem rejects call often while person dialed me once and waits for connection. When this happens too often (e.g. 5 times in ~40 seconds) modem drops from the network: it is not rebooted, but cell icon shows disconnect and reconnection in a while. I do not know if this is some problem within modem or this is some safety mechanics of mobile network.

Biktorgj, 2022-12-27:
The reason why it vibrates is because things crash when you get a second call notification while the call is already in progress. As far as I know that functionality is still missing in gnome-calls (that is precisely the reason for having the callwait* functions).

Auto Reject and Auto Ignore are both hacks around the bug and do more or less the same: when a second (or third, or nth) call notification comes from the baseband, the modem userspace hijacks that message and doesn't forward it to the PinePhone. That way, neither ModemManager or oFono get to know there's a second call, and nothing makes the calls app crash. If you use Auto Reject, the userspace will additionally send a disconnect request to the baseband without notifying the PinePhone, so the caller gets a busy tone.

The behaviour you're describing (modem drop if a lot of consecutive calls are made), might have a very valid reason. All the communication between the modem and the host is based on "transaction IDs", that is, each message has its own ID, and those IDs are incremented by 1 on each part of a transaction (i.e. call indication: 0 -> answer request: 1 -> answered ACK: 2). While a new transaction will always start from 0, if there's a rapid succession of messages of the same kind transaction IDs will go up, but since we're eating some of those messages, ModemManager might get confused because it figured out there's some message missing, and will drop and reconnect the modem.

Jab2870, 2023-01-16:
> @neovalis Sending `callwait auto ignore` in the chat should make the modem send you a text message when you have a call waiting, but avoid passing that information to ModemManager, making the phone _not_ ring when you get that second call. Can you try that?

Could you please clarify how exactly I should send `callwait auto reject`? Is that an AT command?

bircoph, 2023-01-16:
> Could you please clarify how exactly I should send `callwait auto reject`? Is that an AT command?

This is not AT command, this is SMS command. See https://github.com/the-modem-distro/pinephone_modem_sdk/blob/kirkstone/docs/SMS_INTERFACE.md

Just send this message to internal sms number +223344556677 . This way one can interact with the modem via any sms application.