the-modem-distro/pinephone_modem_sdk #130: eg25-manager can't set AGPS on the CFW(?)

SantX27, 2022-07-07:
Hi, this is more of question, since I can't prove that this works on my modem with the stock firmware.
(Manjaro Unstable has some problems with ADB)
So, some time ago, this commits were merged to eg25-manager:
https://gitlab.com/mobian1/devices/eg25-manager/-/merge_requests/15
https://gitlab.com/mobian1/devices/eg25-manager/-/merge_requests/27
https://gitlab.com/mobian1/devices/eg25-manager/-/merge_requests/29
Which supposedly, should automagically send the modem the necessary AGPS data.
However, I've never seen this automatic procedure work with the community firmware.
Can somebody confirm this issue? I'm a bit afraid of escalating this to the Mobian guys.
Again, sorry if I didn't test more.

satcom886, 2022-07-12:
I have that same exact behavior with `0.6.7` and Manjaro Unstable.

SantX27, 2022-07-27:
I have tested the stock modem firmware and the AGPS procedure works there, getting through all 9 upload steps.

Biktorgj, 2022-07-28:
Has anyone tried with 0.6.8? Seems to work for me... If it doesn't, can you post eg25manager log?

SantX27, 2022-07-28:
I have installed 0.6.8 yesterday, here is the log starting from today, when I turned on the phone.

```
lug 28 10:48:13 miria eg25-manager[2962]: Opening config file: /usr/share/eg25-manager/pine64,pinephone-1.2.toml
lug 28 10:48:13 miria eg25-manager[2962]: Starting modem...
lug 28 10:48:14 miria eg25-manager[2962]: Executed power-on/off sequence
lug 28 10:48:14 miria eg25-manager[2962]: taking systemd sleep inhibitor
lug 28 10:48:14 miria eg25-manager[2962]: oFono vanished from D-Bus
lug 28 10:48:14 miria eg25-manager[2962]: ModemManager appeared on D-Bus
lug 28 10:48:14 miria eg25-manager[2962]: inhibitor sleep fd is 21
lug 28 10:48:34 miria eg25-manager[2962]: Response: [RDY]
lug 28 10:48:34 miria eg25-manager[2962]: taking systemd sleep inhibitor (blocking)
lug 28 10:48:34 miria eg25-manager[2962]: Response: [+CPIN: READY]
lug 28 10:48:34 miria eg25-manager[2962]: Executed soft sleep sequence
lug 28 10:48:34 miria eg25-manager[2962]: inhibitor block fd is 20
lug 28 10:48:34 miria eg25-manager[2962]: Response: [+QUSIM: 1]
lug 28 10:48:34 miria eg25-manager[2962]: Executed soft sleep sequence
lug 28 10:48:35 miria eg25-manager[2962]: Response: [+CFUN: 4]
lug 28 10:48:35 miria eg25-manager[2962]: Executed soft sleep sequence
lug 28 10:48:35 miria eg25-manager[2962]: Response: [+QIND: SMS DONE]
lug 28 10:48:35 miria eg25-manager[2962]: Executed soft sleep sequence
lug 28 10:48:36 miria eg25-manager[2962]: Response: [+QIND: PB DONE]
lug 28 10:48:36 miria eg25-manager[2962]: Executed soft sleep sequence
lug 28 10:48:43 miria eg25-manager[2962]: Rescheduling upload since modem isn't online yet, in 30s
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager object `/org/freedesktop/ModemManager1/Modem/0' added
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp.ProfileManager' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Firmware' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Time' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Voice' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: Adding new modem `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: Executed soft wake sequence
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QGMR
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Messaging' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp.Ussd' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Signal' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Location' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: Adding new modem with location capabilities `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Simple' found on object `/org/freedesktop/ModemManager1/Modem/0'
lug 28 10:48:46 miria eg25-manager[2962]: Response: [0.6.8\r\n\r\nOK]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QDAI?
lug 28 10:48:46 miria eg25-manager[2962]: Response: [OK]
lug 28 10:48:46 miria eg25-manager[2962]: Got a different result than expected, changing value...
lug 28 10:48:46 miria eg25-manager[2962]: Expected: [1,1,0,1,0,0,1,1]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QDAI=1,1,0,1,0,0,1,1
lug 28 10:48:46 miria eg25-manager[2962]: Response: [OK]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QCFG="risignaltype"
lug 28 10:48:46 miria eg25-manager[2962]: Response: [+QCFG: "risignaltype","physical"\r\n\r\nOK]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QCFG="ims"
lug 28 10:48:46 miria eg25-manager[2962]: Response: [+QCFG: "ims",1,0\r\n\r\nOK]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QCFG="apready"
lug 28 10:48:46 miria eg25-manager[2962]: Response: [+QCFG: "apready",1,0,500\r\n\r\nOK]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QCFG="urc/ri/ring"
lug 28 10:48:46 miria eg25-manager[2962]: Response: [+QCFG: "urc/ri/ring","pulse",120,1000,5000,"off",1\r\n\r\nOK]
lug 28 10:48:46 miria eg25-manager[2962]: Successfully sent command: AT+QCFG="urc/ri/smsincoming"
lug 28 10:48:47 miria eg25-manager[2962]: Response: [+QCFG: "urc/ri/smsincoming","pulse",120,1\r\n\r\nOK]
[... cut ...]
lug 28 15:58:40 miria eg25-manager[2962]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Simple' found on object `/org/freedesktop/ModemManager1/Modem/2'
lug 28 16:08:39 miria eg25-manager[2962]: Response: [+QGPSURC: "xtradataexpire",0,"2022/07/27,02:00:00"]
lug 28 16:08:39 miria eg25-manager[2962]: GNSS assistance data upload already in process (3/9)
lug 28 16:18:39 miria eg25-manager[2962]: Response: [+QGPSURC: "xtradataexpire",0,"2022/07/27,02:00:00"]
lug 28 16:18:39 miria eg25-manager[2962]: GNSS assistance data upload already in process (3/9)
```

Biktorgj, 2022-08-03:
Stupid question, but did you set up the udev rules as recommended in the settings doc?

Does this only happen in Manjaro?

SantX27, 2022-08-03:
I did set it up as described in SETTINGS.md, but it got overwrote by the latest eg25-manager update, so I'm back at the default ruleset. 

```
cat /usr/lib/udev/rules.d/80-modem-eg25.rules 

ACTION!="add", GOTO="eg25_end"
SUBSYSTEM!="usb", GOTO="eg25_end"
DRIVER!="usb", GOTO="eg25_end"
ENV{DEVTYPE}!="usb_device", GOTO="eg25_end"

# Default attributes values
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/control}="auto"
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/autosuspend_delay_ms}="3000"
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/wakeup}="enabled"
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/persist}="0"

# If running the stock firmware, stop processing here
ATTRS{serial}!="community_fw", GOTO="eg25_end"

# power/control needs to be "on" for the community-maintained firmware
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/control}="on"

# Special trick for the PinePhone Pro: set power/persist to 1 *only* with the community FW
# We can identify the PPP by looking for the string "pinephone-pro" in the device tree "compatible" property
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", PROGRAM=="/bin/grep pine64,pinephone-pro /proc/device-tree/compatible", ATTR{power/persist}="1"

LABEL="eg25_end"

```
As for other distros, I have tried on postmarketOS a while ago and the issue was still there, I'm going to flash an SD with it and see if it happens there as well.

SantX27, 2022-08-03:
I have now tested postmarketOS, enabling logs as described in
https://wiki.postmarketos.org/wiki/Modem#Getting_eg25-manager_logs
and setting the udev rules as described in SETTINGS.md

The upload works, but not on the first time, I need to restart eg25-manager to make it go through. Maybe eg25-manager never receives the response to AT+QGPS?, stalling the upload process.

I will try restarting eg25-manager on Manjaro, maybe it works there too. 

```
** Message: 08:32:12.512: Opening config file: /usr/share/eg25-manager/pine64,pinephone-1.2.toml
** Message: 08:32:12.542: Starting modem...
** Message: 08:32:13.642: Executed power-on/off sequence
** Message: 08:32:13.645: ModemManager vanished from D-Bus
** Message: 08:32:13.645: oFono vanished from D-Bus
** Message: 08:32:14.080: ModemManager appeared on D-Bus
** Message: 08:32:15.040: taking systemd sleep inhibitor
** Message: 08:32:15.059: inhibitor sleep fd is 19
** Message: 08:32:35.064: Response: [RDY]
** Message: 08:32:35.065: taking systemd sleep inhibitor (blocking)
** Message: 08:32:35.082: inhibitor block fd is 20
** Message: 08:32:35.103: Response: [+CPIN: READY]
** Message: 08:32:35.103: Executed soft sleep sequence
** Message: 08:32:35.117: Response: [+QUSIM: 1]
** Message: 08:32:35.117: Executed soft sleep sequence
** Message: 08:32:35.446: Response: [+CFUN: 4]
** Message: 08:32:35.447: Executed soft sleep sequence
** Message: 08:32:35.668: Response: [+QIND: SMS DONE]
** Message: 08:32:35.668: Executed soft sleep sequence
** Message: 08:32:36.898: Response: [+QIND: PB DONE]
** Message: 08:32:36.898: Executed soft sleep sequence
** Message: 08:32:42.917: Rescheduling upload since modem isn't online yet, in 30s
** Message: 08:32:46.548: ModemManager object `/org/freedesktop/ModemManager1/Modem/0' added
** Message: 08:32:46.549: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp.ProfileManager' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.549: ModemManager interface `org.freedesktop.ModemManager1.Modem.Firmware' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.549: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.549: ModemManager interface `org.freedesktop.ModemManager1.Modem.Time' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.549: ModemManager interface `org.freedesktop.ModemManager1.Modem.Voice' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.549: ModemManager interface `org.freedesktop.ModemManager1.Modem' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.549: Adding new modem `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.750: Executed soft wake sequence
** Message: 08:32:46.750: Successfully sent command: AT+QGMR
** Message: 08:32:46.750: ModemManager interface `org.freedesktop.ModemManager1.Modem.Messaging' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.750: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp.Ussd' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.750: ModemManager interface `org.freedesktop.ModemManager1.Modem.Signal' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.751: ModemManager interface `org.freedesktop.ModemManager1.Modem.Location' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.751: Adding new modem with location capabilities `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.751: ModemManager interface `org.freedesktop.ModemManager1.Modem.Simple' found on object `/org/freedesktop/ModemManager1/Modem/0'
** Message: 08:32:46.768: Response: [0.6.8\r\n\r\nOK]
** Message: 08:32:46.768: Successfully sent command: AT+QDAI?
** Message: 08:32:46.786: Response: [OK]
** Message: 08:32:46.786: Got a different result than expected, changing value...
** Message: 08:32:46.786: Expected: [1,1,0,1,0,0,1,1]
Response: [OK]
** Message: 08:32:46.786: Successfully sent command: AT+QDAI=1,1,0,1,0,0,1,1
** Message: 08:32:46.805: Response: [OK]
** Message: 08:32:46.806: Successfully sent command: AT+QCFG="risignaltype"
** Message: 08:32:46.833: Response: [+QCFG: "risignaltype","physical"\r\n\r\nOK]
** Message: 08:32:46.834: Successfully sent command: AT+QCFG="ims"
** Message: 08:32:46.849: Response: [+QCFG: "ims",1,0\r\n\r\nOK]
** Message: 08:32:46.850: Successfully sent command: AT+QCFG="apready"
** Message: 08:32:46.876: Response: [+QCFG: "apready",1,0,500\r\n\r\nOK]
** Message: 08:32:46.877: Successfully sent command: AT+QCFG="urc/ri/ring"
** Message: 08:32:46.905: Response: [+QCFG: "urc/ri/ring","pulse",120,1000,5000,"off",1\r\n\r\nOK]
** Message: 08:32:46.905: Successfully sent command: AT+QCFG="urc/ri/smsincoming"
** Message: 08:32:46.933: Response: [+QCFG: "urc/ri/smsincoming","pulse",120,1\r\n\r\nOK]
** Message: 08:32:46.934: Successfully sent command: AT+QCFG="urc/ri/other"
** Message: 08:32:46.961: Response: [+QCFG: "urc/ri/other","off",1,1\r\n\r\nOK]
** Message: 08:32:46.962: Successfully sent command: AT+QCFG="urc/delay"
** Message: 08:32:46.978: Response: [+QCFG: "urc/delay",0\r\n\r\nOK]
** Message: 08:32:46.978: Successfully sent command: AT+QCFG="urc/cache"
[... 189 lines cut ...]
** Message: 08:42:43.180: Response: [OK]
** Message: 08:42:43.181: Executed soft sleep sequence
[... cut ...]
** Message: 08:42:49.478: Successfully sent command: AT+QGPSXTRADATA="RAM:xtra2.bin"
** Message: 08:42:49.494: Response: [OK]
** Message: 08:42:49.494: Executed soft sleep sequence
** Message: 08:42:49.494: GNSS assistance data upload finished
** Message: 08:42:49.495: GNSS assistance upload step (7/9): re-enabling GNSS through ModemManager
** Message: 08:42:49.512: GNSS assistance upload step (8/9): re-enabling GNSS through AT+QGPS
** Message: 08:42:49.512: GNSS assistance upload step (9/9): finished
```

SantX27, 2022-08-03:
Took a while, but after 4 restarts it did the trick!
Posting only the last 2 for the sake of brevity.

```
ago 03 11:03:43 miria eg25-manager[4526]: Opening config file: /usr/share/eg25-manager/pine64,pinephone-1.2.toml
ago 03 11:03:43 miria eg25-manager[4526]: Starting modem...
ago 03 11:03:44 miria eg25-manager[4526]: Executed power-on/off sequence
ago 03 11:03:44 miria eg25-manager[4526]: taking systemd sleep inhibitor
ago 03 11:03:44 miria eg25-manager[4526]: ModemManager appeared on D-Bus
ago 03 11:03:44 miria eg25-manager[4526]: oFono vanished from D-Bus
ago 03 11:03:44 miria eg25-manager[4526]: inhibitor sleep fd is 21
ago 03 11:04:06 miria eg25-manager[4526]: Response: [RDY]
ago 03 11:04:06 miria eg25-manager[4526]: taking systemd sleep inhibitor (blocking)
ago 03 11:04:06 miria eg25-manager[4526]: inhibitor block fd is 20
ago 03 11:04:06 miria eg25-manager[4526]: Response: [+CPIN: READY]
ago 03 11:04:06 miria eg25-manager[4526]: Executed soft sleep sequence
ago 03 11:04:06 miria eg25-manager[4526]: Response: [+QUSIM: 1]
ago 03 11:04:06 miria eg25-manager[4526]: Executed soft sleep sequence
ago 03 11:04:06 miria eg25-manager[4526]: Response: [+CFUN: 4]
ago 03 11:04:06 miria eg25-manager[4526]: Executed soft sleep sequence
ago 03 11:04:06 miria eg25-manager[4526]: Response: [+QIND: SMS DONE]
ago 03 11:04:06 miria eg25-manager[4526]: Executed soft sleep sequence
ago 03 11:04:07 miria eg25-manager[4526]: Response: [+QIND: PB DONE]
ago 03 11:04:07 miria eg25-manager[4526]: Executed soft sleep sequence
ago 03 11:04:14 miria eg25-manager[4526]: Rescheduling upload since modem isn't online yet, in 30s
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager object `/org/freedesktop/ModemManager1/Modem/1' added
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp.ProfileManager' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Firmware' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Time' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Voice' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:16 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:16 miria eg25-manager[4526]: Adding new modem `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: Executed soft wake sequence
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QGMR
ago 03 11:04:17 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Messaging' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Modem3gpp.Ussd' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Signal' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Location' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: Adding new modem with location capabilities `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Simple' found on object `/org/freedesktop/ModemManager1/Modem/1'
ago 03 11:04:17 miria eg25-manager[4526]: Response: [0.6.8\r\n\r\nOK]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QDAI?
ago 03 11:04:17 miria eg25-manager[4526]: Response: [OK]
ago 03 11:04:17 miria eg25-manager[4526]: Got a different result than expected, changing value...
ago 03 11:04:17 miria eg25-manager[4526]: Expected: [1,1,0,1,0,0,1,1]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QDAI=1,1,0,1,0,0,1,1
ago 03 11:04:17 miria eg25-manager[4526]: Response: [OK]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QCFG="risignaltype"
ago 03 11:04:17 miria eg25-manager[4526]: Response: [+QCFG: "risignaltype","physical"\r\n\r\nOK]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QCFG="ims"
ago 03 11:04:17 miria eg25-manager[4526]: Response: [+QCFG: "ims",1,0\r\n\r\nOK]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QCFG="apready"
ago 03 11:04:17 miria eg25-manager[4526]: Response: [+QCFG: "apready",1,0,500\r\n\r\nOK]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QCFG="urc/ri/ring"
ago 03 11:04:17 miria eg25-manager[4526]: Response: [+QCFG: "urc/ri/ring","pulse",120,1000,5000,"off",1\r\n\r\nOK]
ago 03 11:04:17 miria eg25-manager[4526]: Successfully sent command: AT+QCFG="urc/ri/smsincoming"
ago 03 11:04:17 miria eg25-manager[4526]: Response: [+QCFG: "urc/ri/smsincoming","pulse",120,1\r\n\r\nOK]
[... cut ...]
ago 03 11:07:50 miria eg25-manager[4722]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Location' found on object `/org/freedesktop/ModemManager1/Modem/4'
ago 03 11:07:50 miria eg25-manager[4722]: Adding new modem with location capabilities `/org/freedesktop/ModemManager1/Modem/4'
ago 03 11:07:50 miria eg25-manager[4722]: ModemManager interface `org.freedesktop.ModemManager1.Modem.Simple' found on object `/org/freedesktop/ModemManager1/Modem/4'
ago 03 11:08:45 miria eg25-manager[4722]: Modem is up for 120 seconds and fully ready
ago 03 11:08:45 miria eg25-manager[4722]: dropping systemd sleep block inhibitor
 
```

stacyharper, 2022-08-16:
I relate the same issue on pmos with the vanilla setup from the distro. Got the same udevs and same logs. GNSS assistance upload step stop at (3/9). One time I had the 9/9 but other issues came and I didnt had a fix

stacyharper, 2022-08-16:
For it to works I had to do :

- stop modemmanager
- restart eg25-manager
- wait some time (~20/30s)
- start eg25-manager
- after sometime (with the next scheduled 30s trigger) eg25-manager successfully load GNSS

I still got issues with gpsd… Following the previous step I manage one time to see GNSS satelites on `cgps`. I had to `mmcli -m any --location-enable-gps-unmanaged`. But now I fail to reproduce it…

Biktorgj, 2022-08-31:
I think we all need to stop hogging the AT interface on the modem, and by all I mean both eg25-manager and the firmware. 
When the modem starts, we're all sending and waiting for arbitrary commands to an AT interface that's shared by three different programs requesting different things at the same time, and things fail.

I'm currently trying to finish some other (unrelated to this particular issue) piece of code, but once I'm done with that I will have to go through my code and try to minimize all interactions with the AT interface so it's free for whatever ModemManager or eg25-manager need (currently it uses it to sync date and time and get network indication data)

a13xie, 2023-07-24:
Hi everyone, I just want to confirm - this is still happening on `0.7.4`. 

As far as I can tell, development of `eg25-manager` has kinda dropped off, which is pretty unfortunate :/

bemoody, 2023-09-19:
I tried making the following horrible kludge of a patch to eg25-manager:

```
--- eg25-manager-0.4.6.orig/src/gnss.c
+++ eg25-manager-0.4.6/src/gnss.c
@@ -16,7 +16,7 @@
 #define BUFFER_SIZE 256
 #define UPLOAD_DELAY_US 25000
 #define UPLOAD_TIMEOUT_S 10
-#define RESCHEDULE_IN_SECS 30
+#define RESCHEDULE_IN_SECS 41
 
 static void gnss_step(struct EG25Manager *manager);
 
```

By delaying the GNSS initialization slightly, this seems to sidestep whatever race condition was happening previously, and it successfully reaches `GNSS assistance upload step (9/9): finished`.

I still haven't yet gotten a fix (maybe because I don't have a clear view of the sky), but I do now see multiple GPGSV and GLGSV lines in the NMEA output, which I didn't see before.

a13xie, 2023-09-19:
@Biktorgj could this please be included in the next release as a temporary fix?

stacyharper, 2023-09-19:
@a13xie This is not up to @Biktorgj to include eg25-manager workaround.

Also, I don't think the RESCHEDULE_IN_SECS is in cause. I tried with 90 and still have the problem. I also tried to implement a timeout behavior in eg25-manager GNSS loading workflow. I then tried to sleep a bit between the ModemManager GNSS disabling, and eg25-manager GPS probing, and still. None of this works.

I think there is a problem in the modem firmware, that is triggered by the chaos from both eg25-manager and ModemManager, initiated by eg25-manager GNSS loading workflow.

a13xie, 2023-09-19:
Oh, sorry, I read this as soon as I woke up and I missed the part where you mentioned this was a fix to eg25-manager and not the FW.