the-modem-distro/pinephone_modem_sdk #134: legal implications of using this firmware

federico-d, 2022-07-18:
It is a bit off topic but I can't find a better place to discuss it.
I got a bit scared reading "Some administrative regions, in the EU and Asia in particular, require the entirety of the modem's firmware to be licensed." on Pine64 wiki (https://wiki.pine64.org/wiki/PineModems).
Do anyone know what are the legal implications of using this firmware in a commercial product?
Can someone expand on what the wiki states?

daniel-santos, 2022-07-20:
This is in inadequate answer IMO, but I still think worth posting.  Anybody with deeper knowledge, please post a resonse!

My SWAG would be that this has to do with a legal agreement between the PineModem manufacturer (which I guess is named Pine Store?) and Qualcomm.  In general, you are bound to whatever agreement you make with the modem/phone manufacturer when you purchase it.  I certainly did not see any legal agreement about not altering the firmware when I purchased it, so I think you're in the clear SO LONG as you purchase this from Pine Store and do not mfg them yourself.

Qualcomm has a very strange business model to me.  Anyway, obviously consult an attorney before you move forward with a commercial venture.

I read that the Qualcomm firmware has gdb, linux, sendmail and other FOSS software.  I wonder if *they* are actually in compliance with the FOSS licenses.

mouffa, 2022-09-07:
I will go more "off-topic", have you ever considered that very soon you may be forced to carry a Google or Apple phone or even a bracelet or implanted version in order to collect your social credit and be able to eat zhe bugz (soylent green)???

Biktorgj, 2022-09-08:
IMPORTANT WARNING: I AM NOT A LAWYER

There are three vendors involved here: Pine64, Quectel and Qualcomm. On one side, Qualcomm and Quectel have an agreement where Quectel is able to modify the source code provided by Qualcomm for their device customization.

The agreement between Pine64 and Quectel will probably put a limit on the modifications they can do themselves to the firmware provided by Quectel, but my assumption is that they just don't want trouble regarding regulatory bodies (FCC etc.)

One thing is to provide the option to do as you wish with an equipment that may transmit in protected bands, another thing entirely is to directly overwrite your OEM provided firmware to something made by some weirdo on the internet :) (even if the weirdo does provide the sourcecode).

@daniel-santos: Yes, Quectel is routinely violating the GPL with their firmware. Starting with the linux kernel, for which the version they provide is incomplete and doesn't match *any* published kernel build in any of their firmware versions, to any userspace application that uses a free license.

They could provide the source code for all the opensource bits used in their firmware, since all that is already provided by Qualcomm (you just need to go to Sierra's site, which makes a very similar module, and download all the opensource stuff in source code, with the proprietary bits prebuilt). But they just don't care.

xnopasaranx, 2022-09-08:
> I will go more "off-topic", have you ever considered that very soon you may be forced to carry a Google or Apple phone or even a bracelet or implanted version in order to collect your social credit and be able to eat zhe bugz (soylent green)???

someone hasn't been wearing their tinfoil hat...

mouffa, 2022-09-08:
@xnopasaranx I can safely assume that your FOSS-ness mentality is compatible with taking closed-source and blank Package Insert Sheet injections

xnopasaranx, 2022-09-08:
> @xnopasaranx I can safely assume that your FOSS-ness mentality is compatible with taking closed-source and blank Package Insert Sheet injections

since I am part of the communist jewish new world order, yes you can safely assume that I am vaccinated. Whatever that has to do with Free Open Source Software (that is what FOSS means btw.) eludes me, but I am sure you will enlighten me regardless... might I suggest however, that you take your hot takes on the future of humanity to some obscure message board, rather than cluttering a development issue tracker (tipp: github is for people developing software, FOSS or proprietary, it is not meant for your very important opinions, that is what twitter is for)?

daniel-santos, 2022-09-09:
> I will go more "off-topic", have you ever considered that very soon you may be forced to carry a Google or Apple phone or even a bracelet or implanted version in order to collect your social credit and be able to eat zhe bugz (soylent green)???

Please take this shit to 4chan or some place else.

daniel-santos, 2022-09-09:
> someone hasn't been wearing their tinfoil hat...

Please don't feed the trolls.  Beautiful responses (I liked the "communist jewish new world order" bit), but please don't fee the trolls.

Does github not have a mechanism to report somebody or have them banned from a project?  We don't need this crap clogging up our sane discussions.

daniel-santos, 2022-09-09:
@Biktorgj Ah, thank you for the information!  Yes, my company will be going through FCC regulatory testing again soon -- it's quite expensive.  It would make sense that they don't want to be attached to an FCC violation, but I'm pretty certain that, legally, there's nothing they can do about a **customer** of Pine altering the firmware.

But your points are very important.  Even if they don't get certified, it's good if somebody with the proper equipment thoroughly tests them to be reasonably certain they aren't interfering with some other band(s).

mouffa, 2022-09-09:
@daniel-santos I am just repeating what the Mainstream Media and your government tells you, that you have to control you carbon footprint with China-like surveillance, nothing to do with 4chan, just World Economic Forum and Klaus Schwab, why are you triggered so much hearing to the daily news ? what are you doing here searching for open software ? don't you trust Apple, Qualcomm, Windows and Google ? is contact tracing and surveillance so bad ?

mouffa, 2022-09-09:
> @Biktorgj Ah, thank you for the information! Yes, my company will be going through FCC regulatory testing again soon -- it's quite expensive. It would make sense that they don't want to be attached [...]

do you really believe that the Law Enforcement of your country can not know that you are altering the firmware and can not do something about it ?

mouffa, 2022-09-09:
> > someone hasn't been wearing their tinfoil hat...
> 
> @xnopasaranx Please don't feed the trolls. Beautiful responses (I liked the "communist jewish new world order" bit), but please don't fee the trolls.
> 
> Does github not have a mechanism to report somebody or have them banned from a project? We don't need this crap clogging up our sane discussions.

I would be very happy to be banned from such places.

pizdjuk, 2022-10-10:
> https://stackoverflow.com/questions/4200800/in-bash-how-do-i-bind-a-function-key-to-a-command

somebody already demanded them? FSF has a lawyer department for this.

GNUtoo, 2023-04-22:
A question here is what is really the modem? To me the EG-25 looks a lot like a smartphone. It even has partitions similar to Android with a boot, recovery, modem, system partitions. Inside it has a Qualcomm SOC that has "computers" (CPU, RAM, peripherals, etc) that run GNU/Linux and other "computers" that run the modem software.

So as I understanding it, "the modem" is constituted of what is in the ADSP partition, the core(s) that runs that code, the radio peripherals attached to these cores, etc.

So it could be that Pine64 doesn't want to take legal risks and to try to explain in a court of law that vision and see if they win or not. Or it could be that they also used an EG-25 because it was already certified and that modifying it or doing a smartphone with a system-on-package/chip that does only modem (like something that would only have the ADSP and that would interface with the A64 directly) would need more (re)-certifications. Or maybe it's for another reason.

As I understand it laws and regulations for selling products are also more strict than for free software after-market third party modifications. For instance devices manufacturers might have to do some certifications for WiFi while the Linux kernel doesn't. Long time ago, Atheros published code for doing certification for ath9k if I recall well, so that code has some uses.

So in our cases a way to understand the risk would be to try to look information on regulations that applies to after-market firmware, see if we didn't miss anything as for what constitute the modem, etc. Many years ago, the SFLC probably had that kind of documentation for at least WiFi, though I'm not sure if it is still up to date, and nowadays it might be better to try to collaborate with the FSF instead.

Also if there is a way to install this "pinephone_modem_sdk" without having to redistribute nonfree software (like what is in the ADSP partition) it might also help limiting legal risks as we would need some permission for that, while we don't need permissions to use the ADSP partition content that is already there.

And limiting legal risks could help a lot having distributions being able to simply redistribute distributions like the "pinephone_modem_sdk".

Note that I'm also not a lawyer and I don't have the full picture so I might miss some things.

mouffa, 2023-04-22:
@GNUtoo the licenses should concern the module and not some abstract notion like the "modem", the term "module" should include every characteristic of the device and especially the identification like the IMEI

the legislation is extremely complicated and practically you can only know when someone comes after you like in the case of the patents, if they want to come after you for some reason they will find something to claim, for the time being for example noone seems to be after Rob Braxman for  advertising that he changes the IMEIs to make the phones work in the USA

so beyond the typical regulations like the emission power, the frequencies and the royalty fees which are settled, the safe choice is to not touch the module

all the built-in backdoors that those companies are mandated to build for interception should mostly reside in the ADSP firmware where everything concerning the 4G implementation resides and that firmware can probably run checks to the rest of the module to locate any kind of interventions, in the case of Pinephone the USB protects you from access to the rest of the phone so they should only be able to see the traffic

even the Quectel manuals are characterised as confidential and are only intended for the manufacturers, you shouldn't be able to run any command like the `+QOPS` for example

an alternative implementation would be building a chipset running OSMOCOM part of which could run on the SoC but as far as I know OSMOCOM is limited in simple GSM because every implementation of 4G is patented and all this design and implementation would demand money, time and probably many other legal problems