the-modem-distro/pinephone_modem_sdk #135: pinephone manjaro kde plasma error dev/ttyUSB2 IS BUSY

Pkos767676, 2022-07-26:
when i run this command `sudo echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2` i get an error that /dev/ttyUSB2 permision denied i run chmod -R 777 /dev/ttyUSB2 AND THEDN I RUN THEprevious command i get an error that /dev/ttyUSB2 IS BUSY

xnopasaranx, 2022-07-26:
I don't think it is smart to change permissions on a serial device (/dev/ttyUSB2), you shouldn't have to do that. The instructions tell you to run the flashall script in a root terminal, not by using sudo. I am not sure what you are trying to do exactly (ADSP?) but try entering a root shell with "sudo su" before.

Biktorgj, 2022-07-26:
If you wrote the command exactly as you put itit's impossible for it to work.
You're running "echo" as root but the end of the pipe as a normal user

Try running this (exactly):
`sudo 'echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2'`

The other option would be to just run the echo command directly as root instead of calling it via sudo.

Pkos767676, 2022-07-26:
> I don't think it is smart to change permissions on a serial device (/dev/ttyUSB2), you shouldn't have to do that. The instructions tell you to run the flashall script in a root terminal, not by [...]

The problem is when I make one call with the pinephone after the call it freezes and I try to find a solution I try to install a driver for the modem in order to find solution

calbasi, 2022-08-02:
> If you wrote the command exactly as you put itit's impossible for it to work. You're running "echo" as root but the end of the pipe as a normal user
>
> Try running this (exactly): `sudo 'echo -ne [...]

This command is not running for me (PP+Mobian):

>  sudo: echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2: no s'ha trobat l'ordre

Where "no s'ha trobat l'ordre, in Catalan, means: the command has not been found...

By the way, the old command (should it be updated with the new one?) is here:

https://github.com/the-modem-distro/pinephone_modem_sdk/blob/honister/docs/SETTINGS.md

airtower-luna, 2022-08-02:
`sudo 'echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2'` isn't working because `sudo` ends up interpreting the whole string as the executable to call (which obviously doesn't exist). You'd need to run the string through a shell to run the redirection as root:

```
sudo sh -c 'echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2'
```

Biktorgj, 2022-08-03:
> This command is not running for me (PP+Mobian):
>
>
> Where "no s'ha trobat l'ordre, in Catalan, means: the command has not been found...
>
> By the way, the old command (should it be updated with the new [...]

AT+CAT can be used to enable the Custom Alerting Tone, not to enter fastboot :)

In any case, half of the document should be removed once all distros update eg25-manager, since it should now take care of setting everything up for you.

If you open your udev file and it looks sort of like [this](https://gitlab.com/mobian1/eg25-manager/-/merge_requests/52/diffs?commit_id=34472a5cff6d6c8caa6e14ec894027e7cf20ed73) you probably don't need to touch it anymore

calbasi, 2022-08-03:
> 

> AT+CAT can be used to enable the Custom Alerting Tone, not to enter fastboot :)

But this is exactly the command you put in your comment, here: https://github.com/the-modem-distro/pinephone_modem_sdk/issues/135#issuecomment-1195625616

calbasi, 2022-08-03:
> In any case, half of the document should be removed once all distros update eg25-manager, since it should now take care of setting everything up for you.
>
> If you open your udev file and it looks [...]

Thanks! It seems the new version for eg25-manager is merged at 0.4.4, a week ago. But Mobian is still using, nowadays, version 0.4.3-1

But I suppose version 0.4.4 can take much to be in Mobian, so I just wait until there.

Maybe could be interesting edit the Settings doc, as you suggested, to remove the obsolete part: https://github.com/the-modem-distro/pinephone_modem_sdk/blob/honister/docs/SETTINGS.md

Zapeth, 2022-08-24:
> If you open your udev file and it looks sort of like [this](https://gitlab.com/mobian1/eg25-manager/-/merge_requests/52/diffs?commit_id=34472a5cff6d6c8caa6e14ec894027e7cf20ed73) you probably don't need to touch it anymore

Out of curiosity, is this a hardware quirk that requires the power setting to be permanently enabled, or just a temporary workaround until the root cause has been determined/fixed? Considering the battery capacity isn't that great to begin with, I'd like to avoid changes like these unless they are absolutely required.

bfra2373, 2022-12-05:
@calbasi Did you look if you are part of the usb group? I had some issue with that because manjaro's user are not part of the USB group by default hence not been able to sudo ttyUSB

calbasi, 2022-12-08:
> 

I don't think so. In fact, there is no usb group on Mobian (at least at /etc/group)

Silverdev2482, 2024-06-18:
I have tried to do sudo echo > file, the way that actually works is echo "" | sudo tee file.