the-modem-distro/pinephone_modem_sdk #138: Do not work

Emilio2801, 2022-08-11:
Manjaro plasma on SD. The command after sudo su 
echo -ne "AT+EN_CAT\r"  > /dev/ttyUSB2
do not work in my pinephonepro

jg110, 2022-08-11:
Just using `echo` didn't work for me either. Send the AT command with minicom instead, that's what I did and now I get an emulated alerting tone.

Emilio2801, 2022-08-12:
Thanks.I am not Linux expert. Do not know how to us minicom
Can you point me to some web page? Thanks

-------- Mensaje original --------

Emilio2801, 2022-08-12:
Could be this sentence to be used?

sudo minicom -b 115200 -D /dev/ttyUSB2

-------- Mensaje original --------

Biktorgj, 2022-08-12:
As root, run `minicom -D /dev/ttyUSB2`

Then, type `ATZ` and press enter (doesn't matter if you dont see anything), it should answer with an "OK"

After that, send AT+EN_CAT and hit enter to enable custom alerting tones

Emilio2801, 2022-08-12:
Ohh! Many thanks. How do I send AT+EN_CAT?

-------- Mensaje original --------

Biktorgj, 2022-08-12:
Same as with ATZ, just type it in Minicom and press ENTER :)

Emilio2801, 2022-08-12:
Thanks I will try during the day. I am going to teach my grandchildren s to ride a bike. Have a nice day

-------- Mensaje original --------

Emilio2801, 2022-08-12:
sudo su
Passwordinicom -D /dev/ttyUSB2
And
minicom:cannot open /dev/ttyUSB2:No such file or directory
Note:pinephonepro running Manjaro plasma on the SD

-------- Mensaje original --------

Emilio2801, 2022-08-12:
Ohh!
I did it in the eMMC and after installing minicom and executing the first command I write ATZ and the answers was OK but when writing AT+EN_CAT it halt in AT and could not continue. I press, anyway all the rest but see nothing
I exit with Ctrl+A
I did it in the SD because I can have a conversation with another phone but in the eMMC with PostmarketosOS The call arrives to the other phone but no conversation is possible.
How can I get the minicom work in the SD?
Regards

-------- Mensaje original --------

Emilio2801, 2022-08-13:
Hi
reading I found setserial that I installed with yay (not possible with pacman) In the SD. Then I execute
Sudo setserial -g /dev/tryUSB[01] and get
/dev/ttyUSB0
/dev/ttyUSB1
Could be one of them be used instead of ../ttyUSB2 for minicom to work and get the dial tone from the SD?
Regards
Emilio

-------- Mensaje original --------

Emilio2801, 2022-08-13:
Finally I could close minicom via ssh as Ctrol +A and then Q it was much easier but even when I got the OK for the two sets
ATZ
Ok
AT+EN_CAT
Ok
I still do not here the Dialing tone. Many thanks for your help and teach.
Nice weekend
Emilio

-------- Mensaje original --------

jg110, 2022-08-17:
According to [this](https://forum.manjaro.org/t/on-the-pinephone-how-to-communicate-with-the-modem-with-at-commands/60884) you should have a /dev/EG25.AT on Manjaro that points to the modem, although according to that thread the modem is on /dev/ttyUSB2.

List the devices under /dev (do `ls /dev`) and see if there's anything in there that looks relevant.

If nothing else you could put Mobian on a microSD card and use it to send the AT commands, since then you'll definitely have the modem under /dev/ttyUSB2.

Emilio2801, 2022-08-17:
Thanks. I will investigate. But it happen yesterday that the Manjaro plasma on SD do not wake up with tow boot as everyday so I need to flash a new SD. I did nothing but installed 'satellite" via flatpak and used it twice with no result. I do not understand why it happened. I changed the SD with an Arch phosh and wake up very well. But I have postmarketOS plasma on the eMMC and I cannot even have a conversation with the other end. As I am learning I do not care much but , sometimes, it is a little bored but I appreciate a lot the affords of programmers to help us so I should keep on going and I will do with your help as well. Thanks

-------- Mensaje original --------

Emilio2801, 2022-08-28:
In /dev. Yes I have
EG25.AT
EG25G.MODEM
EG25.JMRA

What to do now?
Regards

-------- Mensaje original --------

Emilio2801, 2022-08-28:
Sorry
EG25.NMEA
Last one

-------- Mensaje original --------

Emilio2801, 2022-08-28:
I also installed last Mobian version on the SD and the telephone works very well. Dial tone and voice with the other end. Also mmcli for position and gpsd but not geoclue

-------- Mensaje original --------

airtower-luna, 2022-08-28:
`/dev/EG25.AT` should be a symbolic link to the device to send `AT` commands to. That way you don't have to guess which of the `/dev/ttyUSB*` ones is the right one. :slightly_smiling_face:

Emilio2801, 2022-10-11:
Finally I could do
minicom -D /dev/ttyUSB2 on my SD and. Type ATZ
Ok
AT+EN_CAT
Ok

But I am not exiting minicom well . I have the virtual keyboard and
Press Ctrl +A and then X but The terminal remains black.Something don't work and I thing it is exiting minicom , as no result. I need to exit minicom well to get results, I guess. Thank you. I keep on going

-------- Mensaje original --------

Biktorgj, 2022-10-15:
Doesn't matter if you exit minicom correctly or not, if the command is sent and you get an OK you're good to go.

Also, the custom alert tone setting is persistent, if you enable it once it should stay like that until you change it or flash the stock firmware.
I've also been asked to add this as a setting in the chat, so I'll probably add it soon there too

closed 2022-10-15 by Biktorgj

Emilio2801, 2022-10-15:
Thanks. Closed

-------- Mensaje original --------

Emilio2801, 2023-03-07:
Good morning again. I installed pipewire and I have telephones calls in both directions and can hear voice but the other end receive the voice but has also a strong noice. If I could make you a phone call from Barcelona, Spain you could have a first hand impression and guide me with the problem in my PinephonePro. Would it be possible for you to help?
Note
I updated to 0.8.4
Regards Emilio

-------- Mensaje original --------