the-modem-distro/pinephone_modem_sdk #140: Add more documentation on how to get started modifying the firmware.

CoderThomasB, 2022-08-30:
I wanted to try and modify the custom firmware but I could find where the file with the code that determines how the modem works. There are quite a few markdown docs on how to build and install but not many on how the internal modem software works and how to modify it and add new features.

Biktorgj, 2022-08-31:
The 'glue' that makes the modem work as a modem (and messages, time syncing etc.) is Openqti: https://github.com/the-modem-distro/meta-qcom/tree/kirkstone/recipes-modem/openqti

Anything specific I can help with?

CoderThomasB, 2022-09-07:
I've updated the issue with a bit more detail on what this issue is about.

Biktorgj, 2022-09-09:
Hope this helps a bit:
[image]

As I said in the previous response, `openqti` is what makes everything work. Once the system is booted and the ADSP firmware is loaded, all the "modem" things are handled by it. 

If you're diving into the source, `openqti.c` is that spawns the proxies, AT command server, scheduler etc. 
If you wanted, for example, to add a new command to the SMS functionality, you should go to `commands.c` and `commands.h` 
That is the command parser, `rmnet_proxy` will intercept any incoming message to its fake number and redirect it to `sms.c`, which will then go to `command.c` to check if whatever you sent it matches some available command. If it does, then `command.c` will do whatever it has to do and add a message to the queue indicating the result of the command (or the command not found message).

On a practical level, if you wanted to add a "Hello world" command, you'd start adding it [here](https://github.com/the-modem-distro/meta-qcom/blob/1f5dc4a66fa16653bd366425b76a8af257645079/recipes-modem/openqti/files/inc/command.h#L59), and then add a case for it in this [switch](https://github.com/the-modem-distro/meta-qcom/blob/1f5dc4a66fa16653bd366425b76a8af257645079/recipes-modem/openqti/files/src/command.c#L954), following the same pattern as the other commands there.

I should do a cleanup of all that too...