the-modem-distro/pinephone_modem_sdk #141: I cannot call or receive calls

aka0kuro, 2022-09-08:
I have a problem with calls and I would like you to help me to solve it. Thanks in advance. 
When I call from my mobile to another device, the call hangs up, directly. And when they call me, the call is busy all the time, although I have mobile data and receive SMS. 
I use pinephone with postmarketos, but it has also happened to me with manjaro. 
I don't know where the problem may lie...

Biktorgj, 2022-09-09:
You were never able to place calls or is that something that only happens with the custom firmware?

aka0kuro, 2022-09-15:
No. Before, I could make calls, but this problem happens since I put the custom firmware

aka0kuro, 2022-10-02:
I have already found the solution, it is because I had 4G alone, by putting 3G the problem is solved.

closed 2022-10-15 by Biktorgj