the-modem-distro/pinephone_modem_sdk #142: Modem not showing over USB

cristiioan, 2022-09-08:
Hello,
After what I think it was a battery drain issue, my modem stopped showing up over USB. It is fully accessible over Serial, but it won't connect over USB(I did try AT+RESETUSB, and it didn't work). As of now, I have been trying without luck the following solutions:
1. Receive an SMS worked a few times (while I was away, so I was not able to do), but it stopped.
2. Get into fastboot mod: worked once, but after reinstall the issue is still present
3. Short the Test pins to get in EDL – No luck, after multiple tries, but 2 times, it did make the modem to show up(but not in EDL, now I can't replicate that aswell)
4. Use ADB to get in EDL - not possible due to not showing over USB.

From what I have read, it looks like I have a data corruption issue. LSUSB doesn't show any modem, and eg25-manager and Modem Manager print some unhelpful error messages(I did search a bit on the internet). The most consistent one (showing each boot) it the device not supported one(aka no plugin was found).

To test if it is a systemd related issue, I will do tomorrow some tests using Postmarket OS as well. Currently, I did all my test on Mobian, with some done on Arch Linux ARM as well(SMS one and fastboot)

Related to upgrades, I did an upgrade to the modem 2-3 weeks ago, from 0.67(or what was before) to 0.6.8. I noticed the issue last Sunday, but I'm not sure if it happened before.

For more detailed log files, please contact me and I will provide them (modem side over serial, and anything else). I didn't add them here since they are a mess (over 50 reboots in the last week).

It would be helpful to have my modem functional next week.

Thanks,
Cristian

Biktorgj, 2022-09-09:
Okay, I'm trying to make a picture of what's going on but I'm missing some important information.

I'll start by saying that the fact that the serial port still answers to AT commands and replies to something like AT+RESETUSB might hint more to a hardware fault in the Pinephone mainboard than anything else, but let's not get ahead of ourselves.

If you're still running the custom firmware, we need to see the kernel boot log from the Pinephone with all killswitches set to ON. There the modem should briefly appear entering into fastboot mode at least. Getting into EDL is tricky without a soldering iron and a lot of patiente so you could have missed it.

Can you post here a `dmesg` from a fresh boot? (don't care much about the OS, it's ok if you boot from Mobian, pmOS, Arch... whatever you want, but if it's a fresh install better to discard other possible issues related to eg25manager for example

cristiioan, 2022-09-09:
Hello @Biktorgj,
If it is a hardware issue, what could have caused it? It didn't happen before, so I'm surprised!
Below is the dmesg that I got on SXMO Postmarket OS. It is a fresh boot. 
[log.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9536264/log.txt)

closed 2024-08-10 by susanthenerd