the-modem-distro/pinephone_modem_sdk #147: Echo cancellation

bunglesss, 2022-09-27:
Hi,

Did I read somewhere the closed firmware had echo cancellation, and the new open source version hasn't got it? Is this correct?

Biktorgj, 2022-09-29:
Yes, the closed firmware has some voice calibration profiles that are loaded on boot, that tune up audio, and seem to do some basic echo cancellation and automatic gain control in calls, and that's not available in the opensource firmware (all in that is closed source and besides the problems derived of actually making that work there's the licensing issue where we might not be legally able to redistribute those calibration files)

bunglesss, 2022-09-29:
Interesting, thanks for sharing that I only ask because so far I've found the PPP unusable for making phone calls, everyone who calls me can hear themselves echo back.

I did try with the stock firmware and found it unusable there too.

I'm not sure how much of this is echo cancellation, and how much these mic boost/volume settings I've seen posted about, and how much about the hardware ... not sure if I saw something saying the mic might physically be too close to the speaker or something.

CodePhase, 2022-10-06:
@bunglesss I don't think ti's the speaker being too close to the mic on the PPP.  I have been experimenting with this issue for quite some time (on Mobian) and found the "other side echo" persists even when the PPP is face down on my couch, which should at least reduce the echo if that was the cause, but it doesn't.

I found that if I mute the microphone on the PPP that the echo disappears.  I don't know if this is an issue with the modem firmware, the ADSP, or ALSA (UCM profiles).  My guess is that the issue is something either in the audio codec used in the kernel or the UCM profile or a combination of both.  Unfortunately I still haven't gotten to the bottom of it and it's the last major complaint I get when using this phone.

I actually created an issue with Mobian about this [here](https://salsa.debian.org/Mobian-team/devices/pinephonepro-support/-/issues/9), which was imported from the previous Mobian Gitlab instance to the current one.

closed 2022-10-07 by bunglesss