the-modem-distro/pinephone_modem_sdk #149: GPS logging during standby directly on modem firmware

boroli, 2022-09-30:
GPS logging, during outdoor activities like hiking and cycling, can be done e.g. with OsmAnd on Android or MapOut on iOS, including when the phone is on aeroplane mode. Then, even after a full day of logging your GPS track, the phone battery is typically still charged 80%.
Even with SailfishX, on a Sony Xperia X, GPS tracking was working with Android apps reliably, but as waydroid currently [doesn't support GPS](https://github.com/waydroid/waydroid/issues/226) locations yet, this is not even an option to think about.

GPS information on the PinePhone can be accessed when ModemManager is in Aeroplane mode via [gpsd](https://wiki.mobian-project.org/doku.php?id=location#gpsd-based-applications) A very primitive way of GPS logging could look like this via gpsd:
```
while true; do
    date --iso-8601=seconds >> gpslog.txt 
    upower -i /org/freedesktop/UPower/devices/battery_axp20x_battery | grep -i percentage >> gpslog.txt
    gpspipe -r -n20 | grep -i GGA >> gpslog.txt
    sleep 60
done
```
Or even by directly accessing the right USB port:

```
awk -F"," '/GGA/ {print $3,$5}' /dev/ttyUSB1
```

But once the phone is suspended, in order to save battery, also those options stop to work. If the phone isn't suspended, then the battery drains within 4-6h or so, which also doesn't seem appropriate.

Is there a possibility to implement GPS logging directly in the modem firmware, and activate/deactivate it on a user needs base?
It seems that eg25-manager doesn't handle the GPS [by default](https://gitlab.com/mobian1/eg25-manager/-/issues/2) currently, if I understood that commit correctly.

The PinePhone has a bad GPS reception in general. Without [hardware modification](https://forum.pine64.org/showthread.php?tid=11680&pid=97011#pid97011), a GPS signal in covered forests and narrow gorges might not be available - but that's an additional problem.

Biktorgj, 2022-10-08:
Hi
If I remember correctly, prior to suspend ModemManager stops the GNSS engine in the modem, precisely to save battery life.
You could access the GPS port directly inside the modem and have a script run in there storing the raw nmea data if you want, but be aware that battery life will generally be quite affected by that.

You can sniff around the GPS data by enabling it with an AT command (AT+QGPS=1) and then, inside an adb session, opening /dev/smd7.

I can't do anything with ModemManager turning GPS off though, as, from the modem's perspective, it's impossible to know if that was done by a user or by a daemon by itself...

boroli, 2022-11-11:
Thanks for the directions, that seems to be pretty straightforward. Activating gps and logging into the adb shell:

```
sudo sh -c 'echo -ne "AT+QGPS=1\r" > /dev/ttyUSB2'
adb shell
```
The following small script should do the gps logging: 
```
#!/bin/sh
date +%FT%T%z > /var/log/gps.log
head -n 20 /dev/smd7 | grep -m 1 GGA >> /var/log/gps.log
sleep 60

while true; do
    date +%FT%T%z >> /var/log/gps.log 
    head -n 20 /dev/smd7 | grep -m 1 GGA >> /var/log/gps.log
    sleep 60
done
```
As the filesystem of them modem is read only, I put the script into `/run/mount/`. Is there any better solution?

And the gps.log can then just be pulled as:
`adb pull /var/log/gps.log`

What would be the best way to have a daemon running on the modem?
Would it be more advised to setup a (minutely) cron job?

Biktorgj, 2022-11-24:
You can enable persistent logging on the modem, and then "/persist" will be writable.
The easiest thing you can do to automate it and leave it running all the time would be:
1. Put your script in /etc/init.d (mount the system partition readwrite first with `mount -o remount,rw /`
2. Make a symlink from your script to rc3: `ln -s /etc/init.d/your_script /etc/rc3.d/S99_your_script`
(S99 means Start, in position 99)
Alternatively, you could add it to `/etc/inittab`, just like openqti does, so it respawns if it dies for some reason, but given it's a shell script I don't think it would need much restarting