the-modem-distro/pinephone_modem_sdk #156: 0.7.0 reports ADSP version as "unknown"

airtower-luna, 2022-10-15:
After upgrading from 0.6.8 to 0.7.0 both the `version` SMS command and `AT+GETADSPVER` report the ADSP firmware version as "Unknown ADSP Firmware". I'm using ADSP firmware version 30.006, modem firmware 0.6.8 identified it correctly.

I see this in openqti.log:

```
 [227.0961] I handle_atfwd_response: AT CMD: +GETADSPVER
[227.0969] E /lib/firmware/image/modem.b12[227.0973] E read_adsp_version: Error calculating the MD5 for your firmware () 
[227.0977] W read_adsp_version: Could not detect your ADSP firmware! 
```

Biktorgj, 2022-10-15:
You're totally right, already fixed for 0.7.1 (https://github.com/the-modem-distro/meta-qcom/commit/449c90d250b12e20b20fa306c6134d4957a736d9)

I changed a symlink but forgot to change the path in openqti...
Sorry for that!

airtower-luna, 2022-11-24:
I can confirm it works with 0.7.1, thank you! :smiley_cat:

closed 2022-11-24 by airtower-luna