the-modem-distro/pinephone_modem_sdk #158: i keep getting what i assume to be modem noise

bgkillas, 2022-10-17:
on postmarketos sxmo i get texts from this number 3999(im in canada) with 2 dashes as the content of the message implying mms i have recieved dozens of messages at seemingly random increments. happened for as long as i can remember, so months
modem ver: 0.7.0
[sms.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9799166/sms.txt)

xnopasaranx, 2022-10-18:
I am sure this cannot be noise. Have you heard of the short code program? A way for businesses and government to register 4-6 digit numbers... maybe you have been subscribed to some newsletter/advertisement crap and need to simply opt out? Try sending STOP back to that number and see if you can get a result back... also research short code programs online, not sure I am supposed to leave a link here that is somewhat marketing related. They have different command sets to interact. Apparently Bell has a short code program and there is a canadian short code program as well. Just guessing here by the way, hope it helps.

Biktorgj, 2022-11-24:
That's 100% coming from the network. The reason I don't know. The way the network informs the modem that there's a MMS to download (and the modem to the mms stack) should be transparent and shouldn't end up in the inbox (unless that's an SXMO thing), so I agree with @xnopasaranx, that seems to be either some broken ad/informational message from some company malfunctioning (or in an encoding SXMO doesn't understand), or some spurious message coming from somwehere.
The modem only sends stuff from +223344556677 :)
If you receive various of those messages a day, have you checked if they also show up the same with Plasma/Phosh?
Only thing that I can think of is that the message could be encoded in non-gsm7 and sxmo may be having trouble decoding that message (hence the two dashes). Chatty (Phosh) does support non-latin alphabets and doesn't seem to have issues decoding) UTF-8, so maybe worth a try?

bgkillas, 2022-11-24:
it sent 2 more messages after i sent "STOP" at the regular times then it stopped so either "STOP" worked or the number just does not get used anymore idk

closed 2022-11-24 by bgkillas