the-modem-distro/pinephone_modem_sdk #159: [Mobian] Won't Connect

qstambaugh, 2022-10-21:
I'm currently running a Pinephone convergence model with Mobian Bookworm installed. I installed the open-source firmware (version 0.7.0) for the modem via fwupd. Ever since, I haven't had a working connection to the T-Mobile network in the US. I haven't tried any other cellular providers.

If I reboot my phone, sometimes it will connect to the cellular network and I can access the internet through it for a few seconds. I've tried troubleshooting it, trying different ADSP version, and even restoring to the original firmware. I don't know what I did, but it's not working even with the original firmware flashed back to it. Is anyone able to help?

[dmesg-modem.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9842836/dmesg-modem.txt)
[dmesg-pinephone.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9842837/dmesg-pinephone.txt)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9842838/eg25-manager.log)
[modemmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9842839/modemmanager.log)
[networkmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9842840/networkmanager.log)

zimward, 2022-10-23:
Looking at your pp dmesg (and the empty modem dmesg) it seems like your modem isn't booting. Have you tried [these instructions](https://github.com/the-modem-distro/pinephone_modem_sdk/blob/kirkstone/docs/TROUBLESHOOTING.md#distributions-running-megis-modem-power-kernel) ? I had the same happening after manually installing it (the last update from 0.6.8 to 0.7.0 via fwupd worked without any trouble for me, also on mobian bookworm. so its probably not mobians fault)

qstambaugh, 2022-10-24:
Unfortunately, it's still having the same issue. I've ran through them, rebooting the modem and it shows up fine as:

`Bus 002 Device 008: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem`

I did leave the modem enables a couple of days ago and it did connect every few minutes for a few seconds. I'll get a set of logs tomorrow with it running longer

qstambaugh, 2022-10-24:
Okay, I have more substantial logs now.

[dmesg-modem.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9855379/dmesg-modem.txt)
[dmesg-pinephone.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9855380/dmesg-pinephone.txt)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9855382/eg25-manager.log)
[modemmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9855383/modemmanager.log)
[networkmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9855384/networkmanager.log)
[openqti.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9855385/openqti.log)

qstambaugh, 2022-10-25:
It works fine in my Pixel 3a. The APN is set to fast.t-mobile.com on both devices

qstambaugh, 2022-10-29:
It was randomly working reliably for a few minutes, so I grabbed logs in that time frame

[dmesg-modem.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9892613/dmesg-modem.txt)
[dmesg-pinephone.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9892615/dmesg-pinephone.txt)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9892616/eg25-manager.log)
[modemmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9892617/modemmanager.log)
[networkmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9892618/networkmanager.log)
[openqti.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9892619/openqti.log)

Biktorgj, 2022-10-29:
It seems it's being rejected all the time until randomly it gets an IPv6 address, the rest of the times it get ` [modem0/bearer2] verbose call end reason (2,231): [internal] ip-version-mismatch`

I'd like to check if the correct lte profile is being activated first. Please install minicom / picocom /microcom, and connect to the modem's serial port. For example:
`minicom -D /dev/ttyUSB2`
Then send the following command: `AT+QMBNCFG="List"`

You should get something like this:
```
+QMBNCFG: "List",0,1,1,"ROW_Generic_3GPP",0x0501081F,202112292
+QMBNCFG: "List",1,0,0,"VoLTE-ATT",0x0501033C,202105171
+QMBNCFG: "List",2,0,0,"hVoLTE-Verizon",0x05010141,202111151
+QMBNCFG: "List",3,0,0,"Sprint-VoLTE",0x05010205,202010201
+QMBNCFG: "List",4,0,0,"Commercial-TMO_VoLTE",0x05010505,202112101
+QMBNCFG: "List",5,0,0,"Telus-Commercial_DO",0x0580F601,202105061
+QMBNCFG: "List",6,0,0,"Commercial-SBM",0x05011C18,202004211
+QMBNCFG: "List",7,0,0,"Commercial-DT",0x05011F1C,202201041
+QMBNCFG: "List",8,0,0,"Reliance_OpnMkt",0x05011B38,202003251
+QMBNCFG: "List",9,0,0,"TF_Germany_VoLTE",0x05010C1B,202004151
+QMBNCFG: "List",10,0,0,"TF_Spain_VoLTE",0x05010CFA,202106171
+QMBNCFG: "List",11,0,0,"Volte_OpenMkt-Commercial-CMCC",0x05012071,202201063
+QMBNCFG: "List",12,0,0,"VoLTE_OPNMKT_CT",0x050113FC,202201063
+QMBNCFG: "List",13,0,0,"CU-VoLTE",0x05011508,202201063
+QMBNCFG: "List",14,0,0,"Telstra-Commercial_VoLTE",0x0580079E,202112221
+QMBNCFG: "List",15,0,0,"Commercial-KDDI",0x0501071D,202011192
+QMBNCFG: "List",16,0,0,"Commercial-DCM",0x05010D17,202111091
+QMBNCFG: "List",17,0,0,"Commercial-SKT",0x05012715,202103121
+QMBNCFG: "List",18,0,0,"Commercial-KT",0x05012C0D,202108241
+QMBNCFG: "List",19,0,0,"Commercial-LGU",0x05012626,202108241
+QMBNCFG: "List",20,0,0,"Commercial-USCC",0x0504FC40,202011111
+QMBNCFG: "List",21,0,0,"Optus-Commercial_VoLTE",0x05800CA2,201910241
+QMBNCFG: "List",22,0,0,"STC_Saudi_VoLTE",0x0501FE01,201912231
+QMBNCFG: "List",23,0,0,"Commercial-Rogers",0x05018821,202108021
```
Can you post the output of the command here?

Besides that, it's also possible that the modem is trying to get an ipv4 but keeps getting rejected because it only supports ipv6. Does it show signal bars, even if it has no data and no voice capability? 

Last thing this could be related to (that I can think of) is that it's trying to connect to a network it is not allowed to. What is the behaviour if you tell the modem to *only* use 4G LTE, and disable 2G/3G support? (In Phosh that would be Settings -> Mobile Network -> 4G Only instead of 2G,3G,4G)

qstambaugh, 2022-10-29:
Here's the feedback

`+QMBNCFG: "List",3,0,0,"Sprint-VoLTE",0x05010205,202010201
+QMBNCFG: "List",4,1,1,"Commercial-TMO_VoLTE",0x05010505,202112101
+QMBNCFG: "List",5,0,0,"Telus-Commercial_DO",0x0580F601,202105061
+QMBNCFG: "List",6,0,0,"Commercial-SBM",0x05011C18,202004211
+QMBNCFG: "List",7,0,0,"Commercial-DT",0x05011F1C,202201041
+QMBNCFG: "List",8,0,0,"Reliance_OpnMkt",0x05011B38,202003251
+QMBNCFG: "List",9,0,0,"TF_Germany_VoLTE",0x05010C1B,202004151
+QMBNCFG: "List",10,0,0,"TF_Spain_VoLTE",0x05010CFA,202106171
+QMBNCFG: "List",11,0,0,"Volte_OpenMkt-Commercial-CMCC",0x05012071,202201063
+QMBNCFG: "List",12,0,0,"VoLTE_OPNMKT_CT",0x050113FC,202201063
+QMBNCFG: "List",13,0,0,"CU-VoLTE",0x05011508,202201063
+QMBNCFG: "List",14,0,0,"Telstra-Commercial_VoLTE",0x0580079E,202112221
+QMBNCFG: "List",15,0,0,"Commercial-KDDI",0x0501071D,202011192
+QMBNCFG: "List",16,0,0,"Commercial-DCM",0x05010D17,202111091
+QMBNCFG: "List",17,0,0,"Commercial-SKT",0x05012715,202103121
+QMBNCFG: "List",18,0,0,"Commercial-KT",0x05012C0D,202108241
+QMBNCFG: "List",19,0,0,"Commercial-LGU",0x05012626,202108241
+QMBNCFG: "List",20,0,0,"Commercial-USCC",0x0504FC40,202011111
+QMBNCFG: "List",21,0,0,"Optus-Commercial_VoLTE",0x05800CA2,201910241
+QMBNCFG: "List",22,0,0,"STC_Saudi_VoLTE",0x0501FE01,201912231
+QMBNCFG: "List",23,0,0,"Commercial-Rogers",0x05018821,202108021`

It normally doesn't show bars when it's not working. Sometimes it does, but I haven't noticed it that often. I've tried it both in mixed mode and 4G only mode, it doesn't make a difference.

qstambaugh, 2023-01-09:
Sometimes, when it tries to connect (but ends up failing), it will say it's connected to AT&T or other random providers, even though its just setup for T-Mobile and their APN