the-modem-distro/pinephone_modem_sdk #162: Modem not working

Peter2469, 2022-11-04:
Hello,
I updated it to 0.7.0 and since then my modem does not work anymore and doesn't seem ever entirely to authenticate. (Cannot receive phone calls/text messages)

I have attached logs to see if you can see if there is anything wrong with it but if I try with Phosh it seems to also be an issue.

The Sim did work previously and would likely still do and on occasion, modemmanager does pick up the right information but never sets it properly.

[openqti.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9934153/openqti.log)
[networkmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9934154/networkmanager.log)
[modemmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9934155/modemmanager.log)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9934156/eg25-manager.log)
[dmesg-pinephone.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9934157/dmesg-pinephone.txt)
[dmesg-modem.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9934158/dmesg-modem.txt)

Biktorgj, 2022-11-24:
Hi Peter,
ModemManager seems to be crashing on you:
```
Nov 04 01:05:38 pinephone systemd[1]: ModemManager.service: Main process exited, code=dumped, status=6/ABRT
Nov 04 01:05:38 pinephone systemd[1]: ModemManager.service: Failed with result 'core-dump'.
```
The modem seems healthy from the log, it seems to boot find and after 120 seconds it even syncs the time from the network, so everything is "up" in that side. Any pending modemmanager update? Can you get a larger modemmanager log? (to see if that issue repeats all the time)

Peter2469, 2022-11-24:
I am fully up to date on my device and I can tell that the main issue of the sim not being fully "managed" is on repeat with GSM Network Timeout; I have uploaded a log which whilst isn't long shows a repeated issue 
[ModemManager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10083749/ModemManager.log)