the-modem-distro/pinephone_modem_sdk #164: Sending AT command over mmcli broken

boroli, 2022-11-11:
Hi,

since some time I get the following error message, when I try to send AT commands via ModemManager:

```
~/$ sudo mmcli -m any --command=$(echo AT+CCLK=\"`date -u +%y/%m/%d,%H:%M:%S+01`\")
error: command failed: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: Cannot send AT command to modem: operation only allowed in debug mode'
```
or alternatively without sudo

```
~/$ mmcli -m any --command=$(echo AT+CCLK=\"`date -u +%y/%m/%d,%H:%M:%S+01`\")
error: command failed: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unauthorized: PolicyKit authorization failed: not authorized for 'org.freedesktop.ModemManager1.Device.Control''
```
And actually the MM man pages states, that this is only available in debug mode:
```
--command=COMMAND
              Send  an  AT  COMMAND  to  the  given  modem. For example, COMMAND could be
              'AT+GMM' to probe for phone  model  information.  This  operation  is  only
              available when ModemManager is run in debug mode.
```

I tried to find some meaningful information, but google wasn't any helpful in this case. I thought maybe some group id's are missing?!?
```
~/$ id
uid=1000(mobian) gid=1000(mobian) groups=1000(mobian),20(dialout),27(sudo),29(audio),44(video),46(plugdev),104(input),107(render),111(bluetooth),120(feedbackd)
```
In that particular instance, I tried to set the time and tried to send the AT command directly to USB2:

```sudo mmcli -m any --command=$(echo AT+CCLK=\"`date -u +%y/%m/%d,%H:%M:%S+01`\")```

But also this wasn't working.

boroli, 2022-11-11:
After some research I figured out how to start [ModemManager in debug mode](https://modemmanager.org/docs/modemmanager/debugging/):

```
sudo service ModemManager stop
sudo /usr/sbin/ModemManager --debug
```

Then on a second shell (or maybe you can put the ModemManager output in bg), send your AT commands:

```
sudo mmcli -m 0 --command=$(echo AT+CCLK=\"`date -u +%y/%m/%d,%H:%M:%S+01`\")
sudo mmcli -m 0 --command=$(echo AT+CCLK?)
response: '+CCLK: "22/11/11,16:23:58+44"'
```
That seems to have worked. But when I log into the adb shell, it still gives me 1970 date:

```
~$ adb shell
sh-5.1# date
Thu Jan  1 01:01:34 UTC 1970
```
What's the reason for that?
Then back to the first shell, or bring MM into foreground:

```
CTRL+C
sudo service ModemManager start
```

boroli, 2022-11-11:
If I send an AT command directly to USB2, where can I read the response to it, like for this one?
```
sudo sh -c 'echo "AT+CCLK?\r" > /dev/ttyUSB2'
```

airtower-luna, 2022-11-11:
You can't, because nothing in that command stays around to listen for a response. If your distro has `atinout` that's probably the simplest option:

```
echo 'AT+CCLK' | sudo atinout - /dev/ttyUSB2 -
```

Otherwise look at serial terminals (e.g. minicom, picocom, IIRC you can make `screen` work as one, too).