the-modem-distro/pinephone_modem_sdk #166: Mudita Pure

sarahspock, 2022-11-16:
Mudita pure is a dumb phone with an [open source OS](https://github.com/mudita/MuditaOS) and a Quectel EG25-G modem. I am wondering if it is possible to flash the firmware onto this phone instead of the PinePhone. The instructions for flashing the firmware are specific to the linux OS on the PinePhone and as such will not work on the open source muditaOS.

dogtopus, 2022-11-17:
It seems like their OS is not linux based (and why would it). So you might need to find the USB test points for the modem and do it manually. If it's not provided you might be out of luck.

Biktorgj, 2022-11-17:
Mudita OS is based on FreeRTOS, so flashing from the phone itself would probably be impossible.
Furthermore, I'm not even sure if USB is connected to the main SoC at all, it could be wired to only use the serial port so flashing that way would be impossible.
Only chance I could think of is building some FOTA package, but if it breaks or fails your modem would be dead. Totally not recommend it unless you have a recovery method of some sort

Biktorgj, 2022-11-17:
Oh I didn't know that!
I guess making Linux run on that imx SoC to flash the modem would be a little bit overkill though :)