the-modem-distro/pinephone_modem_sdk #168: Can't change sound volume while on a phonecall

bfra2373, 2022-11-24:
Hi there, 

I've been using the latest firmware on a Pinephone and have no control over sound when on a phone call.
I can change it in the VoiceCall.conf file but not while I'm on the phone.
Everything else seems fine.

Arch Danctnix Phosh kernel to 5.18.14

What other info would you need to help me fix that? (I know probably a lot :P)

I did try it on a fresh install just to be certain that my Voicecall.conf was not messed up.

Biktorgj, 2022-11-24:
I guess first things first: While in a call, in Settings -> Sound, the active profile is "PhoneCall" ?

And if it is... if you move the volume sliders from there doesn't it alter the volume from the earpiece and the mic?

Biktorgj, 2022-11-27:
There *has* to be some issue we're not seeing here. I guess the first thing you could do is spawn a root terminal, and, as root, run `alsactl monitor`. Everytime an audio control is triggered, it should spam that terminal with the mixer elements that changed. 
I guess, if volume doesn't respond for you, that the mute button in the call UI doesn't work either? 
It's a disto issue by the way, but we can try to get it sorted out anyway if you want :)

bfra2373, 2022-11-28:
Thank's for the help btw!
I have spawn to log, one is when I change from [mute to unmute](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10103917/alsactl_volume_test.txt) and the other is just a reference when I make a call but [don't touch anything](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10103915/alsactl_untouch.txt). There is no trigger when I change the volume.

[alsactl_untouch.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10103915/alsactl_untouch.txt)
[alsactl_volume_test.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10103917/alsactl_volume_test.txt)

sc0w, 2022-11-28:
The same happens issue here.

The volume works without a call.

In a call the volume slider in the gui works (internal earpiece), but I have always the same volume, it is always the same.

Biktorgj, 2022-11-29:
Seems like something broke lately? Doesn't seem to work in mobian now, but it sure did a week or two ago

Just for the sake of testing, manually setting the control still works for you?

`amixer cset "name='Speaker Playback Volume'" 31`
Values can go from 0 (mute) to 39 (you might break your earpiece)

bfra2373, 2022-11-30:
Running `amixer cset "name='Speaker Playback Volume'" 1`  return `amixer: Cannot find the given element from control default`

bfra2373, 2023-01-09:
On Linux Mobian 5.15-sunxi64, the issue is not present and sound is adjustable during call but updating to latest break it. So it might be  a Kernell issue. 
Running `amixer cset "name='Speaker Playback Volume'" 1` also return `amixer: Cannot find the given element from control default` but sound is adjustable during call

bfra2373, 2023-01-10:
Going through every Manjaro Phosh distribution from kernell 5.15, I found that the issue start's at release beta 26 [beta 26](https://github.com/manjaro-pinephone/phosh/releases).

Alaraajavamma, 2023-06-29:
If you wan't "fast fix" you can use pipewire-pulse and pipewire-media-session 
It works there but not with pulseaudio (and with wireplumber it kind of works but the callaudiod configuration is f*cked so it is not best option as in phone calls)

bfra2373, 2023-08-02:
@Alaraajavamma What would be the long fix?

Alaraajavamma, 2023-08-02:
afaik the end goal would be something like this: 
someone™ will find out how  to make wireplumber work with callaudiod reliable 
That someone™ fixes same time the ucm/alsa/callaudiod/calls issues where for example mic-mute does not work or bluetooth audio does not work out-of-the-box

And after that someone™ will build echo/noice cancellation for this firmware so you can use speaker mode and do not hear static noice when using earpice mode. 

Just hoping that someone™ will be as good and as fast as I have heard :)

sc0w, 2023-09-20:
now, it seems to work again, the volume in calls, can someone confirm?

The issue is fixed for me