the-modem-distro/pinephone_modem_sdk #173: time syncing wrong

bgkillas, 2022-12-01:
while catting /dev/ttyUSB2 to check output i ran

``doas su -c 'echo -ne AT+CCLK=\"22/12/01,00:09:18-24\""\r" > /dev/ttyUSB2'``
``doas sh -c 'echo -ne "AT+CCLK?\r" > /dev/ttyUSB2'``

and it outputs

``OK``
``+CCLK: "24/10/13,10:09:23-24"``

before i tried to set time it was

``+CCLK: "24/10/13,11:08:20-20"``

do i just have to use mmcli? running it in debug mode is so silly

Biktorgj, 2022-12-01:
No need to use ModemManager in debug mode for this, but the command you posted has a typo:
`doas su -c 'echo -ne "AT+CCLK=\"22/12/01,00:09:18-24\"\r" > /dev/ttyUSB2'`

Now, things to keep in mind:
- The last `-24` means the GMT offset in 15 minute increments, so that would be `GMT-6`
- You have a 300 second window from _network up_ to set the clock until openqti gives up. You can easily bypass both the AT command and the time window just using ADB, with `adb shell date`.

OpenQTI doesn't really need the baseband to know the correct time (messages received from the network will still have the correct time set), it just needs the RTC for the userspace to have it so fake SMS's and call recordings come with the correct date (there's two separate clocks in the modem)

bgkillas, 2022-12-01:
still cant set CCLK right but i can set adb shell date correctly at least