the-modem-distro/pinephone_modem_sdk #174: Wrong release in the link

bfra2373, 2022-12-03:
Downloading the link [ADSP Version 01.003.01.003](https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi) from the repo give the wrong firmware. I don't know if it really the wrong firmware altogether or just the wrong version name but `sudo mmcli -m any --command='AT+QGMR'`
return `0.5.1`.

airtower-luna, 2022-12-03:
That is described in the [AT interface documentation](https://github.com/the-modem-distro/pinephone_modem_sdk/blob/kirkstone/docs/AT_INTERFACE.md):

> [7]: QGMR has been replaced in this firmware to report the firmware version instead of the ADSP version, so fwupd doesn't need to query different commands depending if you're running stock or this one

Use `AT+GETADSPVER` to check the ADSP firmware version.

bfra2373, 2022-12-03:
hum.. seem's to work before but now `sudo mmcli -m any --command='AT+GETADSPVER'` return `error: command failed: `GDBus.Error:org.r=freedesktop.ModemManager1.Error.MobileEquipment.Unkown: Unknow error`

Biktorgj, 2022-12-04:
Well, if you're still running 0.5.1 maybe GETADSPVER didn't even exist :)

Any reason why you're 20 releases away from the current one? Did something break?

bfra2373, 2022-12-04:
Well, has I was mentioning in the post.. clicking on the 0.7.1 link return to the 0.5.1 firmware.. that exactly my issue :P

Biktorgj, 2022-12-05:
One thing is the "commnity firmware version" (which is 0.7.1 at the moment of writing) and the other one is the ADSP firmware, which is the proprietary blob from Quectel.

You probably flashed correctly the ADSP part, but you really really should upgrade the rest of the firmware :)

Please, go here: https://github.com/the-modem-distro/pinephone_modem_sdk/releases/tag/0.7.1
And download the file called `package.tar.gz`, unpack it, and then, in a root shell, run the `flashall` script that's included there. The modem should update and reboot, and then you'll be able to run AT+GETADSPVER, or send `version` in the modem chat to check both the firmware and ADSP version

bfra2373, 2022-12-05:
Oh well sorry for that my mistake! I had an old version that took over when I last rewrite the image on my phone.
Issue solved!

So to understand this clearly, the ADSP part is not modified by the community only the package.tar.gz is? should I flash both of them anyway when updating?

closed 2022-12-08 by bfra2373