the-modem-distro/pinephone_modem_sdk #177: I can't access to fastboot on OG Pinephone with sxmo (postmarketos)

castenaso, 2022-12-07:
Hi, i'm on sxmo (pmos), i activated ADBON and after i try to access fastboot (sudo sh -c 'echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2') pinhephone refuse to go on fastboot mode, i mean if i type 'lsusb' no fastboot device showing. Is it wrong command to use on sxmo? is there a better command to use in sxmo?

castenaso, 2022-12-08:
Solved!
With the modem fully booted, and on a root terminal, run 'fastboot oem stay &' (without '). It should stay running in the background waiting for the modem to get into fastboot

Once you have that running, send: echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2 from that same terminal.
There was a bug around those last 0.6.x builds in the bootloader where QFASTBOOT wouldn't kill the timer when entering fastboot mode when requested from the userspace.
If all goes well you should get an [OKAY] from the fastboot oem stay command running in the background. If you do, now you can take as much time as you want to flash the latest version (the bug is fixed in the rest of the builds)

closed 2022-12-08 by castenaso

Snakespeed, 2024-12-20:
Sorry to necro this, but I found I also had to kill -9 the eg25-manager process, and the supervisor daemon for it, before the fastboot command would take.