the-modem-distro/pinephone_modem_sdk #179: SMS commands not working and nothing received

UltraBlackLinux, 2022-12-11:
Hey there,
I just had the same issue as #178. Calling also works, but the sms commands don't seem to work. I also never received the welcome message.
I then tried calling the modem's number, which worked. Then I tried to change the username to see if it would be reflected in the call tts, but it still just called me "admin"
I even tried flashing this firmware again (0.7.1), but that didn't yield any other results too.
The phone I'm trying this on is a PP with danctnix.
In the current state the new firmware behaves like I never flashed it, apart from the calling.
Any ideas what this could be about?
Thanks!

zimward, 2022-12-11:
I just tried to send some commands to the modem via the SMS interface, and the modem receives them properly (for example i get called by it if i send it `call me`) but i dont get an answer from the modem for commands like `memory`. Im running Mobian bullseye on a PPP and firmware version 0.7.1. so it's either a bug in the modem firmware or a bug in chatty. The latter seems more likely to me since i never got a welcome message either after installing it (version 0.6.8) and someone would probably have noticed.

Biktorgj, 2022-12-12:
Not receiving the fake SMS from the modem is usually due to some stuck messages in the modem's memory which confuse ModemManager and make it attempt to retrieve the message from the wrong memory location. 
As I have never been able to reproduce that myself, I'm a bit blind here.
Here's a test release that may (or may not) fix it:
[test_072.tar.gz](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10206261/test_072.tar.gz)

@zimward The issue you describe may not be the same. If it has a delay, probably some message wasn't retrieved in time by modemmanager/ofono and was discarded. The `help` command returns quite a few messages, if one is discarded part of the help could be missing.

Let me know if behaviour changes with this build please

zimward, 2022-12-14:
From my testing with the test release the behavior didn't change. Almost all status commands return nothing and the `load` command occasionally returns a message when directly send after a command with a lot of output (like `help` or `about`), with the other commands it didn't work in my tests. I will get some logs when i have time at the weekend.

I also restarted modemmanager and all commands still got a delay of 1-3 minutes (same goes for rebooting) so i would guess that a not retrieved message isn't the culprit (unless it doesn't get retrieved every time but i also never had a sms that went missing since all i get are 2FA codes)

Biktorgj, 2022-12-15:
@zimward : Let's clean up the modem then and recheck: Please send `AT+WIPESMS` to the AT interface, give it a minute or so and reboot. This will remove any and all stuck/pending messages in modem's EFS, let's see if behaviour changes after that

zimward, 2022-12-15:
@Biktorgj Seems like that that did it. Now all commands reply in an instant.

UltraBlackLinux, 2022-12-15:
AT+WIPESMS seems to have fixed the long message delay on my PPP as well, but the modem still plays dead on my PP.

I was however able to receive messages prior to this, but that appears to be a fluke. It's now no longer working.

devren-y, 2022-12-21:
I am having the same problem, Pinephone Beta Edition, modem firmware 0.7.1.  The modem does respond quickly to commands such as "reboot", but I do not get any feedback if the command is supposed to make a reply, like "version" or "memory".  I just tried the AT+WIPESMS via atinout, and it did not make any difference for me.

It did work for me originally when I installed the firmware on version 0.7.0.

I wonder if it is related to [another issue about timekeeping](https://github.com/the-modem-distro/pinephone_modem_sdk/issues/173) — the one time the modem managed to reply, to an "about" command, the timestamp on the reply texts was 2024-10-27.

Biktorgj, 2022-12-27:
> I wonder if it is related to [another issue about timekeeping](https://github.com/the-modem-distro/pinephone_modem_sdk/issues/173) — the one time the modem managed to reply, to an "about" command, the timestamp on the reply texts was 2024-10-27.

You're using Mint Mobile aren't you? The date won't cause SMS to stop working, though they will show with an invalid date.
Did WIPESMS return an OK?

I'm finishing up a workaround for 0.7.2, where this issue + stuck SMS issues should be fixed. The underlying problem is that when the modem boots, ModemManager asks for all the messages in the baseband's memory, and if there's > 0, things get confusing when mixing fake SMS with real ones. The fix for this is to
1. Lie to ModemManager and tell it there are no pending messages
2. Grab those messages from the baseband
3. Send them as fake SMS from the modem
4. Delete those messages from memory so they stop causing issues

Since ModemManager / oFono are sometimes unable to retrieve certain kind of messages, decoding some message fails, chatty fails, things get stuck waiting for a multipart sms for which some piece never arrives or whatever, what 0.7.2 does is retrieve those stuck messages on the background, decode them and send them back to you as fake SMS (so at least you don't loose them), but since working with SMS is an absolute hell, I still have to do more testing with different character encodings to try to avoid as much data loss as possible, so it's still going to take a bit more time to release it

devren-y, 2022-12-27:
Actually I'm on Postmarket-Phosh, and yes WIPESMS returned OK.

> but since working with SMS is an absolute hell, I still have to do more testing with different character encodings

This reminds me of my old flip phone, which I used for far too long into the smartphone era.  Probably half of all text messages people sent me were unreadable, if they had any kind of emojis or anything but plain ASCII. :)   I am so grateful that someone is putting in the effort to work on this and achieve some sanity!

reivanen, 2023-01-06:
> Actually I'm on Postmarket-Phosh, and yes WIPESMS returned OK.

He asked you if you are on mint mobile (operator) not what distro you are using.

devren-y, 2023-01-06:
OK, in that case I'm on Verizon (USA). :)

bfra2373, 2023-01-30:
Stuck SMS where kind of a nightmare never knowing when my wife sent me a text with an emoji that would fail my SMS :P.
Work's great now on version 7.2!

bfra2373, 2023-01-31:
Well after a couple of day's running 7,2.. Issue seem's to be back! Text message still stop incoming after a while. texting myself would fail has well.

chilledfrogs, 2024-03-06:
Unfortunately running into a variant of this as well, SMS interface just seems wholesale unresponsive to commands (even though I did get a notification of a stuck SMS at one point, ran `AT+WIPESMS`, it returned `OK`, rebooted, no dice), not really sure how to proceed

bfra2373, 2024-03-07:
I had way less issues since the last update. It still happen but most of the time, message get unstuck at reboot.

k8ieone, 2024-06-18:
This started happening to me since the last FW update - 0.8.0. I also happened to flash ADSP 30.202.30.202 right after updating the firmware. When I'd send a command, I would get no response. The modem still got the messages (it reacted to `call me` and reflected username changes, although it took a bit of extra time). `AT+WIPESMS` returned `OK`, but the behavior stayed the same.

Flashing ADSP 30.200.30.200 fixed the problem.

chilledfrogs, 2024-06-21:
Flashed firmware 0.8.0 and ADSP version EG25GGBR07A08M2G_30.202.30.202, now there is a QMI hangup error when I try to send SMSes to the modem, but apart from that SMSes and calls work