the-modem-distro/pinephone_modem_sdk #181: Call hangs up at mailbox

UltraBlackLinux, 2022-12-15:
Hey there,
I just noticed some weird behavior with my modem: Whenever my calls reach someone's mailbox, my modem just hangs up the call a few seconds later.

I found the following in the logs:
modemmanager.log:
```
Dec 15 17:06:13 pinecone ModemManager[3615]: <info>  [modem1/call1] user request to start call
Dec 15 17:06:13 pinecone ModemManager[3615]: <info>  [modem1/call1] call state changed: unknown -> dialing (outgoing-started)
Dec 15 17:06:13 pinecone ModemManager[3615]: <info>  [modem1/call1] call is started
Dec 15 17:06:14 pinecone ModemManager[3615]: <info>  [modem1/call1] call state changed: dialing -> ringing-out (unknown)
Dec 15 17:06:31 pinecone ModemManager[3615]: <info>  [modem1/call1] call state changed: ringing-out -> active (unknown)
Dec 15 17:06:33 pinecone ModemManager[3615]: <info>  [modem1/call1] call state changed: active -> terminated (unknown)
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [base-manager] port ttyUSB0 released by device '/sys/devices/platform/fe3c0000.usb/usb1/1-1'
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [base-manager] port ttyUSB2 released by device '/sys/devices/platform/fe3c0000.usb/usb1/1-1'
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [base-manager] port ttyUSB1 released by device '/sys/devices/platform/fe3c0000.usb/usb1/1-1'
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [base-manager] port ttyUSB3 released by device '/sys/devices/platform/fe3c0000.usb/usb1/1-1'
Dec 15 17:08:30 pinecone ModemManager[3615]: <warn>  Cannot read from istream: connection broken
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [modem1] connection to qmi-proxy for /dev/cdc-wdm0 lost, reprobing
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [base-manager] port wwan0 released by device '/sys/devices/platform/fe3c0000.usb/usb1/1-1'
Dec 15 17:08:30 pinecone ModemManager[3615]: <info>  [base-manager] port cdc-wdm0 released by device '/sys/devices/platform/fe3c0000.usb/usb1/1-1'
Dec 15 17:08:40 pinecone ModemManager[3615]: <info>  [device /sys/devices/platform/fe3c0000.usb/usb1/1-1] creating modem with plugin 'quectel' and '6' ports
Dec 15 17:08:40 pinecone ModemManager[3615]: <warn>  [plugin/quectel] could not grab port ttyUSB3: Cannot add port 'tty/ttyUSB3', unhandled port type
Dec 15 17:08:40 pinecone ModemManager[3615]: <info>  [base-manager] modem for device '/sys/devices/platform/fe3c0000.usb/usb1/1-1' successfully created
Dec 15 17:08:45 pinecone ModemManager[3615]: <warn>  [modem2] couldn't query SIM slots: QMI protocol error (94): 'NotSupported'
Dec 15 17:08:46 pinecone ModemManager[3615]: <warn>  [modem2] couldn't load list of own numbers: Couldn't get MSISDN: QMI protocol error (16): 'NotProvisioned'
Dec 15 17:08:46 pinecone ModemManager[3615]: <info>  [modem2] state changed (unknown -> disabled)
Dec 15 17:08:46 pinecone ModemManager[3615]: <info>  [modem2] state changed (disabled -> enabling)
Dec 15 17:08:46 pinecone ModemManager[3615]: <info>  [modem2] simple connect started...
```

openqti.log:
```
[1151.6121] W handle_call_pkt: Call 1 of 1: Outgoing
[1151.6125] I handle_call_pkt: --> VoLTE
[1151.6129] I handle_call_pkt: --> Call established, mode 2
[1151.8738] I Selected sampling rate: RX: 16000, TX: 16000
[1154.6978] I call_service_handler: Call #0: **********
[1154.6985] I Call ID set: ID 1, number:  **********
[1154.6992] W handle_call_pkt: Call 1 of 1: Outgoing
[1154.6997] I handle_call_pkt: --> VoLTE
[1154.9331] I handle_call_pkt: -->  Disconnecting call...
```

Any idea why I'm getting disconnected? Firmware version is 0.7.1
Thanks!

Biktorgj, 2022-12-27:
Can you post the full log from openqti? and eg25-manager log too?
The modem port is being disconnected, but I don't see and can't imagine _any_ reason for that to happen just because a call ended...

UltraBlackLinux, 2023-04-11:
Oh no I completely forgot about this issue. Here:

https://chonkyrabbit.eu/lettermbox/9pq9c6/ eg25
https://chonkyrabbit.eu/lettermbox/f30voh/ openqti

The modem also ends the call when trying to call a callcenter.