the-modem-distro/pinephone_modem_sdk #182: Modem switches from 4G to 3G during a phone call, can't have a phone call over 4G at all :-(

merinfor2021, 2022-12-18:
When I initiate a phone call, my modem auto-switches from 4G to 3G - despite my `3G, 4G (Preferred)` setting for `Network Mode` . That's a significant issue for me, because in some areas 4G signal is much better than 3G. If the 3G network is unavailable, or I change my `Network Mode` setting to `4G only` - I won't be able to initiate a phone call at all, just get an instant "Call ended".

Similar situation if I receive a phone call: despite I'm on 4G, when the modem sees a phone call is incoming - it auto-switches to 3G. Please, could you help me?

**Hardware:** [Pinephone 1.2](https://wiki.pine64.org/wiki/PinePhone_v1.2) with Quectel EG25-G
**Firmware:** `0.7.1` , but had it on `0.6.8` too and didn't try the earlier firmwares
**ADSP version:** `30.006.30.006` _(I did a full EDL installation - moreover, never had any installation error during these switches)_
**Distro:** PostmarketOS with the latest upgrades _(because no SystemD)_
**Frequencies:** 3G - 2100MHz _(Band 1)_, 4G - 800 MHz _(Band 20)_ / 1800 MHz _(Band 3)_ / 2600 MHz _(Band 7)_

Biktorgj, 2022-12-27:
Modem is unable to negotiate IMS (VoLTE) with your carrier, so it falls back to 3G when doing a call because it's not available.

Which is your carrier?

castenaso, 2023-01-01:
Try to check that
https://forum.pine64.org/showthread.php?tid=12560
its a way to modify profile and choose the better suited profile for your carrier. If you fail using mmcli because you get a strange error regarding debug mode, use minicom. To use minicom (sudo apk add minicom) after install type:
 (minicom -w -D /dev/ttyUSB2) [im not sure about letter -D, and im not sure also on right syntax of the command,  biktorgj knows well minicom], and if i reported well minicom command you can type mmcli -m any --command... blabla, like written on the link above

Piraty, 2023-01-02:
@merinfor2021 check https://wiki.pine64.org/wiki/PinePhone#VoLTE for an appropriate volte profile and find a peer with phone supporting volte to double-check.