the-modem-distro/pinephone_modem_sdk #183: modem appears offline to incoming calls, and won't notify userspace about it

Piraty, 2023-01-02:
Hardware: [Pinephone 1.2](https://wiki.pine64.org/wiki/PinePhone_v1.2) with Quectel EG25-G
Firmware: 0.7.1 (issue happened on all previous versions of this firmware as well as stock firmware too)
ADSP version: EG25GGBR07A08M2G_30.006.30.006/update/NON-HLOS.ubi (issue happened stock EG25GGBR07A08M2G_01.001.10)
Distro: PostmarketOS/edge
___

hi,

issue is present for many month now, so with many firmwares including stock.

i get many complains from other parties that my phone appears disabled. (meaning: network provided voice mail answers to calls)
despite the pinephone showing gsm cov and modem details appear ok.

This happens very often when modem is configured to use "2G, 3G, 4G (preferred)" and registered to 4G and (at least i think) it does not happen on "2G only".
(3G is not available my country)

Any way i can debug this for you ?
sms interface commands `gsm signal`, `net report`, `net stats` didn't seem to be useful at all here.

AndreySV, 2023-01-14:
I've seen this as well on different versions of stock firmware.

carlosgonz0, 2023-01-21:
Seems an IMS issue in MM, just make sure that VoLTE  is enabled in Modem controller and test again, i have a pp but i not using pretty much so not sure how turning ON the VoLTE in Modem in PP.

Piraty, 2023-01-22:
to be clear: VoLTE calls do work in general if the other party supports it as well (verified with a 4G phone).

Biktorgj, 2023-02-14:
@Piraty When the modem is stuck in that state, what does `AT+QENG="SERVINGCELL"` report?