the-modem-distro/pinephone_modem_sdk #187: Document enhancements

alevesely, 2023-01-20:
Hi,
I'm trying to work out why gsm network doesn't work.  Thanks for this site, which looks useful also for gsm ignorant people like me.  However, I have a couple of notes:

**ADSP-CARRIERS.md**
Wouldn't you please spell what does ADSP mean?  Is it digital signal processing something?

Also, why newest versions are low numbered?

And, how do I know what ADSP version I have?

(I found no suggestion for Italian TIM, but I guess there's not much you can do about it.)

**TROUBLESHOOTING.md**
[@kkeijzer's script](https://github.com/Biktorgj/pinephone_modem_sdk/blob/hardknott/tools/helpers/collect_logs) gives errors.  Running with -x, I get:
```sh
[ale@alesmart ~]$ sudo sh -c 'echo -ne "AT+ADBON\r" > /dev/ttyUSB2'
[ale@alesmart ~]$ bin/collect_logs.sh
+ adb shell 'echo Collecting dmesg... > /dev/kmsg'
adb: no devices/emulators found
+ adb shell dmesg
adb: no devices/emulators found
+ adb pull /var/log/openqti.log
adb: error: failed to get feature set: no devices/emulators found
+ sudo sh -c 'echo Collecting dmesg... > /dev/kmsg'
+ dmesg
+ journalctl -u eg25-manager.service -e
+ journalctl -u ModemManager.service -e
+ journalctl -u NetworkManager.service -e
```
(Perhaps the AT+ADBON thing should be in the script itself?)

Anyway, `dmesg-pinephone.txt` is empty.

**My issue**
I haven't installed any new firmware yet.  I dunno what to start with.  NetworkManager's log says:
```log
Jan 20 19:28:35 alesmart ModemManager[3459]: <warn>  Cannot read from istream: connection broken
Jan 20 19:28:35 alesmart ModemManager[3459]: <warn>  [modem7/cdc-wdm0/qmi] Couldn't setup data format: CID allocation failed in the CTL client: endpoint hangup
Jan 20 19:28:35 alesmart ModemManager[3459]: <warn>  [/dev/cdc-wdm0] couldn't detect transport type of port: unexpected port subsystem
Jan 20 19:28:35 alesmart ModemManager[3459]: <warn>  [/dev/cdc-wdm0] requested QMI mode but unexpected transport type found
Jan 20 19:28:35 alesmart ModemManager[3459]: <warn>  Cannot read from istream: connection broken
Jan 20 19:28:35 alesmart ModemManager[3459]: <info>  [base-manager] port cdc-wdm0 released by device '/sys/devices/platform/soc/1c1b000.usb/usb2/2-1'
Jan 20 19:28:35 alesmart ModemManager[3459]: <warn>  [modem7] couldn't start initialization: Couldn't query file info: Error when getting information for file “/dev/cdc-wdm0”: No such file or directory
Jan 20 19:28:35 alesmart ModemManager[3459]: <error> [modem7] fatal error initializing: Fatal error: modem is unusable
```
Then it goes on with modem8, gets IPv4 addresses and happily says ` <info>  [modem8] simple connect state (10/10): all done`.  However, network state is UNKNOWN:
```
[ale@alesmart ~]$ ip addr show wwan0
15: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/none 
    inet 10.58.51.228/29 brd 10.58.51.231 scope global noprefixroute wwan0
       valid_lft forever preferred_lft forever
```

TIA for any hint.

Biktorgj, 2023-02-14:
Hi @alevesely!
If you haven't installed the custom firmware, AT+ADBON won't work. The stock firmware does have an option to enable ADB, but it's more involved (https://xnux.eu/devices/feature/modem-pp.html#toc-unlock-adb-access). 

The ADSP refers to the baseband firmware, a piece of closed source blob that runs inside the modem. The modem has kind of two cpus, the one that controls the RF side of things (the ADSP) and the other, which is what is running linux inside, I also tend to refer to it as "the userspace".

The baseband firmware has its own OS, if you're running the stock firmware from Quectel, the AT command `AT+QMGR` will get you the version you're currently at.

About the version numbers... that's what Quectel provides :) Look at the end of the strings, newer versions end with .003,.004,.006...

The log you pasted from modemmanager shows a usb port reset. I don't know if you're on the PinePhone or the Pro? If you're running the stock firmware on the Pro you're going to have that problem every time the phone wakes up from suspend. On the Pinephone this may happen from time to time, both with stock and the custom firmware, but it's not as bad

The IP address state being unknown doesn't mean much in itself, the loopback interface reports the same too, for example. That value depends on the driver providing a correct value for the state. ModemManager should have logged _everything_ that happened when it requested the IP address, and should have logged the response from the DHCP server on the network. What's the issue there, exactly?

alevesely, 2023-02-14:
Hi,
thank you very much for taking the time to answer me!

Sometimes I seem to understand a little bit more.   When I send AT+QMGR I get ERROR.
That's right after boot:
```sh
Welcome to Manjaro ARM
~~Website: https://manjaro.org
~~Forum:   https://forum.manjaro.org/c/arm
~~Matrix:  #manjaro-arm-public:matrix.org
Last login: Tue Feb 14 19:38:35 2023 from 172.25.197.111
[ale@alesmart ~]$ sudo -s
[root@alesmart ale]# echo 'AT+QMGR' | atinout - /dev/ttyUSB2 -

ERROR
[root@alesmart ale]# echo 'AT' | atinout - /dev/ttyUSB2 -

OK
[root@alesmart ale]# echo 'AT' | atinout - /dev/ttyUSB3 -

OK
[root@alesmart ale]# echo 'AT+QMGR' | atinout - /dev/ttyUSB3 -

ERROR
```

Should I install a recovery version?  If yes, should I do that before or after installing the firmware?

For unknown reason, the GSM network started to work, although I'm unable to test it when I'm connected with wifi.

BTW, in ADSP-CARRIERS.md ther's written *For my use case, I use the latest version, 01.003.01.003.* Which is confusing, since 006 and 004 are listed just underneath.  Why not just install the latest one, anyway?  Shouldn't it be the best?  I downloaded it just to look at it.  Looking for the producer, I only found:
```txt
Copyright (c) 1992-2006 by P.J. Plauger, licensedd
Dinkumware, Ltd. ALL RIGHTS RESERVED.'
```
Hadn't hear of him since the CUJ...