the-modem-distro/pinephone_modem_sdk #188: 0.7.2 is not available on LVFS

pkubaj, 2023-01-31:
I wanted to upgrade with fwupd on a Pinephone, but LVFS has only version 0.7.1.

iambumblehead, 2023-02-04:
https://fwupd.org/lvfs/devices/org.pine64.naoi-os.firmware

airtower-luna, 2023-02-04:
I just checked, 0.7.2 shows up there, but fwupd doesn't offer it after refresh. If I download the update file and try `fwupdmgr local-install` I get:

```
no HWIDs matched 10f570fd-23d3-5079-ae3f-5f4bda888bad|cd16a41d-8ea0-52d1-9e97-4a2c5e94c067|adad873f-33d3-5477-bd5e-0870eca37444
```

iambumblehead, 2023-02-04:
> fwupd doesn't offer it after refresh

same here

Biktorgj, 2023-02-14:
I had some problems getting 0.7.2 validated in LVFS, but now there seems to be some bug in fwupdmgr where it can't correctly process the manufacturer and product for the PinePhone, so it hides all updates for the EG25-G (just filled a bug report for that)
In the meantime, you can either try downgrading fwupd (seems to be working in 1.8.8) or just flash the latest release from github

pentamassiv, 2023-02-28:
It's fixed now. I was able to update to 0.7.2 via fwupd on Arch

Bugsbane, 2023-03-06:
What version of fwupd did you get it working with? @Biktorgj seems to say it should be working with version 1.8.8 but I'm getting this error with 1.8.10 on the PinePhonePro + updated Manjaro...

iambumblehead, 2023-03-06:
@Bugsbane 1.8.11 or 1.8.12 are needed https://github.com/fwupd/fwupd/releases

sammyj85, 2023-04-20:
Now 0.7.2 has been available, but 0.7.4 is not available. (fwupd 1.8.12)

antanasbruzas, 2023-09-20:
> Now 0.7.2 has been available, but 0.7.4 is not available. (fwupd 1.8.12)

+1

hamishmb, 2024-03-18:
Same here, 0.7.4 is still not available on LVFS.