the-modem-distro/pinephone_modem_sdk #192: Problems with donations on KoFi (card payments)

tmczar, 2023-03-02:
I'm trying to buy a coffee on KoFi, using "pay with card" option. Sadly, i'm getting this cryptic error:

[image]

It seems like Stripe account is misconfigured, or something like that (i'm not an expert on internet payments).

Biktorgj, 2023-03-20:
Hi and thanks for the support!
Unfortunately, Stripe doesn't really work for me. I've sent them every document they asked for but they still have me as "unverified". I don't know, they may need a blood sample or something. Everytime I send them some document it works for a few days, then back to "needs verification or payments will stop working". So I just disabled it

tmczar, 2023-03-20:
Thanks for clarification. Adding this info to README may be good idea.

closed 2023-03-20 by tmczar