the-modem-distro/pinephone_modem_sdk #194: Can this firmware support RNDIS/ECM/NCM etc

tinnnysu, 2023-03-18:


Biktorgj, 2023-03-20:
Hi!
I added a small document with some instructions, let me know if they work for you: https://github.com/the-modem-distro/pinephone_modem_sdk/blob/kirkstone/docs/Internal_Networking_Example.md

tinnnysu, 2023-03-21:
I successful enable internal net，but I forget to set apn, no internet from rndis. after that my modem manger gui keep crashing, is there a method to reset this setting?

tinnnysu, 2023-03-25:
rmnet0 doesn't show up on the modem，is there a dial command？

closed 2023-04-03 by Biktorgj

TimelessNL, 2025-06-08:
Hi @Biktorgj,

I was reading your [Internal_Networking_Example.md](https://github.com/the-modem-distro/pinephone_modem_sdk/blob/kirkstone/docs/Internal_Networking_Example.md) to enable local networking on a SIM7600X  (_same chipset_) flashed with 0.8.0. But it is unclear to me where I can find the mentioned chat interface. On the original SIMcom firmware I was able to use `AT+CUSBPIDSWITCH=9011,1,1` to setup a RNDIS connection but that obviously does not work with the alternative firmware. Could you please explain the example a little bit more?