the-modem-distro/pinephone_modem_sdk #195: flashall failures

fdservices, 2023-03-24:
I tried flashall as both user and sudo. As user the result was:

`./flashall`
`Sending AT+QFASTBOOT...`
`< waiting for any device >`
`FAILED (remote: 'unknown command')`
`fastboot: error: Command failed`
`fastboot: error: cannot load 'appsboot.mbn': No such file or directory`
`./flashall: line 8: exit: Failed to flash the bootloader: numeric argument required`

I am running arch linux and android-tools is installed

Can someone tell me where I am going wrong?

fdservices, 2023-03-31:
So I installed the newish firmware using fwupd. Now I have:

`AT+GETSWREV `
`0.7.2`
`OK`
`AT+GETADSPVER`
`EG25GGBR07A08M2G_30.006.30.006`
`OK`

The 0.7.2 is the latest version available to fwupd, but I am not sure if the firmware version number is correct. Can someone confirm please?

airtower-luna, 2023-03-31:
If you mean the ADSP firmware version, that should be fine. There's one newer version (30.007.30.007) [mentioned in the 0.7.3 release notes](https://github.com/the-modem-distro/pinephone_modem_sdk/releases/tag/0.7.3), but if 30.006.30.006 is working fine for you, you don't necessarily have to update. Fwupd won't update the ADSP firmware either way, that's a manual process (as described in the release notes).

fdservices, 2023-03-31:
Not really, the firmware version id does not seem to have changed. Is that correct?

fdservices, 2023-04-01:
The reason I ask is that the firmware version does not seem to have changed, and after the fwupd my calls have frankly stopped working altogether. SMS work fine MMS does not work, calls in are not recognised, calls out fail with

Apr 01 15:47:44 pinephone ModemManager[4891]: <info>  [modem0/call2] user request to start call
Apr 01 15:47:44 pinephone ModemManager[4891]: <info>  [modem0/call2] call state changed: unknown -> dialing (outgoing-started)
Apr 01 15:47:44 pinephone ModemManager[4891]: <info>  [modem0/call2] call is started
Apr 01 15:47:54 pinephone gnome-calls[4300]: Unexpected error response to netlink request while trying to fetch local IP address
Apr 01 15:48:09 pinephone gnome-calls[4300]: Unexpected error response to netlink request while trying to fetch local IP address
Apr 01 15:48:09 pinephone gnome-calls[4300]: SelectMode DBus method invocation failed: Timeout was reached
Apr 01 15:48:09 pinephone gnome-calls[4300]: Failed to select audio mode: Timeout was reached
Apr 01 15:48:17 pinephone ModemManager[4891]: <info>  [modem0/call2] call state changed: dialing -> terminated (unknown)
Apr 01 15:48:17 pinephone callaudiod[4056]: card has no usable source
Apr 01 15:48:17 pinephone callaudiod[4056]: card has no usable sink
Apr 01 15:48:24 pinephone gnome-calls[4300]: Unexpected error response to netlink request while trying to fetch local IP address

Before the fwupd I at least got calls to connect sometimes although the audio was terrible

airtower-luna, 2023-04-01:
The version numbers themselves look okay (and I've used the same combination), I can't comment on what's going on with that call.

Biktorgj, 2023-04-01:
Some issue between Pulseaudio / Pipewire and calls probably : 
```
 Apr 01 15:48:09 pinephone gnome-calls[4300]: Failed to select audio mode: Timeout was reached
```
...
```
Apr 01 15:48:17 pinephone callaudiod[4056]: card has no usable source
Apr 01 15:48:17 pinephone callaudiod[4056]: card has no usable sink
```

If you start a call and manually change the profile from Phosh' Settings, does it work?

fdservices, 2023-04-02:
No real change:

<code>Apr 02 19:30:56 pinephone ModemManager[3495]: <info>  [modem0/call2] user request to start call
Apr 02 19:30:56 pinephone ModemManager[3495]: <info>  [modem0/call2] call state changed: unknown -> dialing (outgoing-started)
Apr 02 19:30:56 pinephone gsd-media-keys[5121]: Unable to get default sink
Apr 02 19:30:56 pinephone gsd-media-keys[5121]: Unable to get default source
Apr 02 19:30:56 pinephone callaudiod[5027]: no available input found!
Apr 02 19:30:56 pinephone gsd-media-keys[5121]: gvc_mixer_card_get_index: assertion 'GVC_IS_MIXER_CARD (card)' failed
Apr 02 19:30:56 pinephone ModemManager[3495]: <info>  [modem0/call2] call is started
Apr 02 19:30:56 pinephone phosh[4995]: gvc_mixer_card_get_index: assertion 'GVC_IS_MIXER_CARD (card)' failed
Apr 02 19:31:00 pinephone gnome-calls[5235]: Unexpected error response to netlink request while trying to fetch local IP address
Apr 02 19:31:05 pinephone kernel: anx7688 0-0028: cc_status changed to CC1 = SRC.Open CC2 = SRC.Open
Apr 02 19:31:05 pinephone kernel: anx7688 0-0028: cc_status changed to CC1 = SRC.Open CC2 = SNK.Power3.0
Apr 02 19:31:06 pinephone kernel: anx7688 0-0028: recv ocm message cmd=0xf2 01 f2 0d
Apr 02 19:31:06 pinephone kernel: anx7688 0-0028: received HARD_RST, idiot firmware is bored
Apr 02 19:31:08 pinephone PackageKit[4639]: get-updates transaction /53_addccccc from uid 1000 finished with success after 10ms
Apr 02 19:31:10 pinephone kernel: anx7688 0-0028: recv ocm message cmd=0xf2 01 f2 0d
Apr 02 19:31:10 pinephone kernel: anx7688 0-0028: received HARD_RST, idiot firmware is bored
Apr 02 19:31:12 pinephone PackageKit[4639]: get-updates transaction /54_eddeebac from uid 1000 finished with success after 10ms
Apr 02 19:31:15 pinephone kernel: anx7688 0-0028: recv ocm message cmd=0xf2 01 f2 0d
Apr 02 19:31:15 pinephone kernel: anx7688 0-0028: received HARD_RST, idiot firmware is bored
Apr 02 19:31:15 pinephone gnome-calls[5235]: Unexpected error response to netlink request while trying to fetch local IP address
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: cc_status changed to CC1 = SRC.Open CC2 = SRC.Open
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: plug irq (cd=0)
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: cable removed
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: power disabled
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: setting vbus_in current limit to 500000 mA
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: disabling vbus_in power path
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: enabling USB BC 1.2 detection
Apr 02 19:31:17 pinephone kernel: anx7688 0-0028: BC 1.2 result: SDP
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: plug irq (cd=1)
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: cable inserted
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: power enabled
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: plug irq (cd=0)
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: plug irq (cd=1)
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: eeprom0 = 0x03
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: fw loaded after 40 ms
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: OCM firmware loaded (version 0x2312)
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: send pd packet cmd=0x00 05 00 32 90 01 26 12
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: plug irq (cd=0)
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: send pd packet cmd=0x01 05 01 2c 91 01 26 16
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: send pd packet cmd=0x02 11 02 00 00 00 ec 00 00 00 00 00 00 00 00 39 00 00 51 77
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: send pd packet cmd=0x03 05 03 00 00 01 ff f8
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: OCM configuration completed
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: status changed to 0x00
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: cc_status changed to CC1 = SRC.Open CC2 = SRC.Open
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: DP state changed to 0x0000
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: VCONN role change to SINK
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: DATA role change requested to UFP
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: cable removed
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: power disabled
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: setting vbus_in current limit to 500000 mA
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: disabling vbus_in power path
Apr 02 19:31:18 pinephone kernel: anx7688 0-0028: enabling USB BC 1.2 detection
Apr 02 19:31:18 pinephone eg25-manager[3308]: Rescheduling upload since modem isn't online yet, in 30s
Apr 02 19:31:20 pinephone PackageKit[4639]: get-updates transaction /55_eedddbbc from uid 1000 finished with success after 18ms
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: plug irq (cd=1)
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: cable inserted
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: power enabled
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: eeprom0 = 0x03
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: fw loaded after 40 ms
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: OCM firmware loaded (version 0x2312)
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: send pd packet cmd=0x00 05 00 32 90 01 26 12
Apr 02 19:31:21 pinephone kernel: anx7688 0-0028: send pd packet cmd=0x01 05 01 2c 91 01 26 16
[... cut ...]
Apr 02 19:32:15 pinephone gnome-calls[5235]: Unexpected error response to netlink request while trying to fetch local IP address
Apr 02 19:32:18 pinephone eg25-manager[3308]: Rescheduling upload since modem isn't online yet, in 30s
</code>

I don't know why flashall does not work for me.

I don't understand <code>eg25-manager[3308]: Rescheduling upload since modem isn't online yet, in 30s</code>

I think the <code>anx7688</code> messages ares due to the power plug disconnecting and reconnecting, which I'm afraid it does frequently.

Biktorgj, 2023-04-03:
You're mixing things up, let's back up a bit please :)

Yes, the anx messages do make it look like your cable is a little loose, but doesn't matter for any of this.
About the `flashall` errors: If it can't find any of the files is because it's not available in the folder where you unpacked the update file. The `flashall` script is only used to flash the entire firmware, if you want to only flash the ADSP, then just use fastboot manually:

With the modem booted, run this (in a root shell):
1. `echo -ne `AT+QFASTBOOT\r' > /dev/ttyUSB2`
2. `fastboot flash modem [PATH_TO_NON_HLOS.ubi] && fastboot reboot`

With that the ADSP will be updated and the modem rebooted afterwards.

As for the audio errors (which don't have anything to do with the ADSP), you have some problem at the distro level: `callaudiod[5027]: no available input found!`, it's not finding the microphone (at least), so switching profiles fail. If you have a microsd nearby I'd recommend you flash a clean distro on it and try to boot from it and check if audio works there, maybe some package broke in your current system?

fdservices, 2023-04-04:
I am afraid I was not making myself very clear. The flashall script fails because appsboot.mbn does not seem to exist in my arch system.

I already updated the ADSP

I think I see the problem with the audio, but I don't know how to fix it. It is some conflict between pipewire and pulseaudio :(

fdservices, 2023-04-04:
pinephone_modem_sdk-0.7.4

I don't see it. I will check again

fdservices, 2023-04-04:
`git clone https://github.com/the-modem-distro/pinephone_modem_sdk.git`

Can you tell me where to find it in the download?

airtower-luna, 2023-04-04:
The binaries aren't in the Git repository. Go to https://github.com/the-modem-distro/pinephone_modem_sdk/releases, pick the version you want, and download the `package.tar.gz` from there.

fdservices, 2023-04-04:
I already tried that with 0.7.4 - as g=far as I can see the contents are identical to the cloned repo.

airtower-luna, 2023-04-04:
> I already tried that with 0.7.4 - as g=far as I can see the contents are identical to the cloned repo.

That means you downloaded the wrong file. You need the `package.tar.gz`, *not* one of the autogenerated source archives.

fdservices, 2023-04-04:
Now I feel really stupid. I found the `package.tar.gz` at the foot of the 0.7.4 page and as you say, it works fine. Thank you for being patient with me.

mikeshenson, 2023-04-08:
I just made a video on how I updated the ADSP version.

https://www.youtube.com/watch?v=XsCpmkE5ttg

Device: PinePhone
OS: DanctNIX Arch Linux Arm
Kernal: Linux danctnix 6.2.9-1-danctnix
Phosh: 0.25.2-1

fdservices, 2023-04-08:
Thank you for the video. I have the same installation as you have Arch 6.2.9-1, Phosh 0.25.2-1 (fresh install)

Biktorgj firmware 
<code>
$ sudo atinout - /dev/ttyUSB2 -
AT+QGMR

0.7.4
OK
</code>

and ADSP ihas been updated
<code>
AT+GETADSPVER

EG25GGBR07A08M2G_30.007.30.007

OK
</code>

The modem is connected:

<code>
$ mmcli -m 0
  -----------------------------------
  General  |                    path: /org/freedesktop/ModemManager1/Modem/0
           |               device id: b526365d8a8c8ebb23bbb0995d36796c19ab76a3
  -----------------------------------
  Hardware |            manufacturer: QUALCOMM INCORPORATED
           |                   model: QUECTEL Mobile Broadband Module
           |       firmware revision: EG25GGBR07A08M2G
           |          carrier config: ROW_Generic_3GPP
           | carrier config revision: 0501081F
           |            h/w revision: 10000
           |               supported: gsm-umts, lte
           |                 current: gsm-umts, lte
           |            equipment id: 867698046452572
  -----------------------------------
  System   |                  device: /sys/devices/platform/soc/1c1b000.usb/usb2/2-1
           |                 drivers: qmi_wwan, option
           |                  plugin: quectel
           |            primary port: cdc-wdm0
           |                   ports: cdc-wdm0 (qmi), ttyUSB0 (qcdm), ttyUSB1 (gps), 
           |                          ttyUSB2 (at), wwu1i4 (net)
  -----------------------------------
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: connected
           |             power state: on
           |             access tech: lte
           |          signal quality: 34% (recent)
  -----------------------------------
  Modes    |               supported: allowed: 2g; preferred: none
           |                          allowed: 3g; preferred: none
           |                          allowed: 4g; preferred: none
           |                          allowed: 2g, 3g; preferred: 3g
           |                          allowed: 2g, 3g; preferred: 2g
           |                          allowed: 2g, 4g; preferred: 4g
           |                          allowed: 2g, 4g; preferred: 2g
           |                          allowed: 3g, 4g; preferred: 4g
           |                          allowed: 3g, 4g; preferred: 3g
           |                          allowed: 2g, 3g, 4g; preferred: 4g
           |                          allowed: 2g, 3g, 4g; preferred: 3g
           |                          allowed: 2g, 3g, 4g; preferred: 2g
           |                 current: allowed: 2g, 3g, 4g; preferred: 4g
  -----------------------------------
  Bands    |               supported: egsm, dcs, pcs, g850, utran-1, utran-4, utran-6, utran-5, 
           |                          utran-8, utran-2, eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, 
           |                          eutran-7, eutran-8, eutran-12, eutran-13, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-38, eutran-39, 
[... 27 lines cut ...]
and SMS works. MMS is not set up.
Calls do not work in or out. Here is the journal for an outgoing call:
<code>
Apr 08 08:57:01 pinephone gnome-calls[4766]: gtk_widget_destroy: assertion 'GTK_IS_WIDGET (widget)' failed
Apr 08 08:57:01 pinephone gnome-calls[4766]: gtk_widget_destroy: assertion 'GTK_IS_WIDGET (widget)' failed
Apr 08 08:57:01 pinephone dbus-daemon[3793]: [session uid=1000 pid=3793] Activating service name='org.gnome.Contacts' requested by ':1.73' (uid=1000 pid=4766 comm="/usr/bin/gnome-calls")
Apr 08 08:57:01 pinephone dbus-daemon[3793]: [session uid=1000 pid=3793] Successfully activated service 'org.gnome.Contacts'
Apr 08 08:57:03 pinephone ModemManager[3438]: <info>  [modem0/call0] user request to start call
Apr 08 08:57:03 pinephone ModemManager[3438]: <info>  [modem0/call0] call state changed: unknown -> dialing (outgoing-started)
Apr 08 08:57:03 pinephone gsd-media-keys[4094]: Unable to get default sink
Apr 08 08:57:03 pinephone gsd-media-keys[4094]: Unable to get default source
Apr 08 08:57:03 pinephone callaudiod[3959]: no available input found!
Apr 08 08:57:04 pinephone ModemManager[3438]: <info>  [modem0/call0] call is started
Apr 08 08:57:27 pinephone geoclue[4382]: Service not used for 60 seconds. Shutting down..
Apr 08 08:57:27 pinephone systemd[1]: geoclue.service: Deactivated successfully.
Apr 08 08:57:38 pinephone ModemManager[3438]: <info>  [modem0/call0] call state changed: dialing -> terminated (unknown)
Apr 08 08:57:38 pinephone gsd-media-keys[4094]: Unable to get default sink
Apr 08 08:57:38 pinephone gsd-media-keys[4094]: Unable to get default source
Apr 08 08:57:38 pinephone callaudiod[3959]: no available input found!
</code>

There are several warnings in the journal some of which look interesting, but I don't know which, if any, are relevant.

.
[journalctl.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/11182739/journalctl.txt)

mikeshenson, 2023-04-10:
I will assume the flashing issue can be closed. 

I think you now have a SIM card not activated or not registered. I will also assume you have not internet. Voice over LTE requires internet. 

From your journalctl.txt, these lines point me to a SIM card issue or provider issue.
```
Apr 08 08:20:47 pinephone NetworkManager[3426]: <warn>  [1680934847.9913] device (cdc-wdm0): retrieving IP configuration failed: modem IP method unsupported
Apr 08 08:21:12 pinephone vvmd[3863]: Could not get modem number!
Apr 08 08:21:13 pinephone mmsdtng[3861]: Could not find settings: Operator ID '208300' with APN 'symamobile.com' not found in service provider database
Apr 08 08:21:13 pinephone mmsdtng[3861]: Your MMS settings are not in the database! Please file a merge request at https://gitlab.gnome.org/GNOME/mobile-broadband-provider-info so they can be added
...
Apr 08 08:39:56 pinephone NetworkManager[3426]: <warn>  [1680935996.1808] device (cdc-wdm0): Activation: failed for connection 'Syma'
```

symamobile.com is not on "PinePhone_Carrier_Support"
https://wiki.pine64.org/wiki/PinePhone_Carrier_Support

"syma" is not on "PinePhone_APN_Settings"
https://wiki.pine64.org/wiki/PinePhone_APN_Settings

fdservices, 2023-04-10:
`I think you now have a SIM card not activated or not registered.`
No, activated and registered, works fine in Android.
`
I will also assume you have not internet. Voice over LTE requires internet.`
No I have internet, VoLTE is not supported by Syma I think?

However the flashall failure has been resolved.

closed 2023-04-10 by fdservices