the-modem-distro/pinephone_modem_sdk #196: Can't disable persistent logging

a13xie, 2023-03-30:
I'm on firmware 0.7.4. Every time I send the SMS to disable persistent logging, `openqti.conf` has the line `persistent_logging=0`, as expected.

However, after a reboot, the value is changed back to `persistent_logging=1` every time.

Here's my full config (with logging disabled):

```
# OpenQTI Config file
# key=value
custom_alert_tone=1
persistent_logging=0
user_name=Alex
modem_name=Modem
signal_tracking=0
signal_tracking_mode=2
signal_tracking_notify_downgrade=0
signal_tracking_notify_cell_change=0
dump_network_tables=0
callwait_autohangup=0
automatic_call_recording=1
sms_logging=0
list_all_bypass=1
allow_internal_modem_connectivity=0
auth_method=0
apn_addr=
apn_username=
apn_password=
```

airtower-luna, 2023-03-31:
I can confirm the problem, also on 0.7.4. After rebooting my phone persistent logging is enabled even though I explicitly disabled it before.

Biktorgj, 2023-04-01:
Could it be triggered because it's not being correctly shut down?
Can you check the value of `cat /persist/.openqti_boot_done`? 
If, when the phone shuts down, the modem is abruptly shut down it's probably not clearing the file (you can just wipe it, disable persistent logging, and send `reboot` to the chat, but I'm curious why it's not being correctly shut down in the first place)

a13xie, 2023-04-01:
If it's because of abrupt shutdowns, I may have caused this. Sometimes my phone starts without the screen working and when I don't have access to a PC to ssh in and reboot the phone, I'm forced to reboot it by either removing the battery or holding the power button.

So maybe the problem is that the file isn't being cleared even on clean shutdowns following an abrupt one?

Just guessing here.

In a boot following a battery removal, the file has `35` in it. After using `reboot` in ADB, the file has `5`.

airtower-luna, 2023-04-01:
I'm currently seeing `101` in `/persist/.openqti_boot_done`. The last boot of the phone was after a hard shutdown (keeping the power button pressed) because the phone was unresponsive. I'd assume that shut the modem off abruptly, too.

airtower-luna, 2023-04-22:
I've confirmed that persistent logging re-activates itself after a clean reboot (of the phone) as well.