the-modem-distro/pinephone_modem_sdk #199: 0.7.4 is not available on LVFS

lgtrombetta, 2023-04-03:
I only see up to version 0.7.2

xnopasaranx, 2023-04-05:
Then why don't you ask on the issue tracker of LVFS? I don't think the modem-firmware devs have any influence on other peoples software

xnopasaranx, 2023-04-05:
Since this issue has been opened numerous times for different release versions, wouldn't it be a good idea to automatically upload firmware releases to LVFS in the CI process? Maybe not until the releases are more stable, not sure about that. See docs here, almost at the end: https://lvfs.readthedocs.io/en/latest/upload.html

liberodark, 2023-07-10:
Hi,

Any news about this ?
Actually LVFS have only the 0.7.2 release.

Best Regards

iambumblehead, 2023-08-01:
a related reddit thread https://www.reddit.com/r/pinephone/comments/157o7ez/modem_firmware_074_and_gnome_firmware/