the-modem-distro/pinephone_modem_sdk #203: Modem disconnects on incoming call

tiol11, 2023-04-30:
Hi and thank you for developing FLOSS firmware for EG25 modem!

Here using OG PP on Mobian bookworm, FW 0.7.4, environment Plasma mobile. Last Wednesday my phone got a call, but nothing happened when I accepted it. After a few seconds, PP rebooted itself.

Of course, I'm not  blaming the modem for the entire device's reboot. But it's strange that a modem disconnects from USB bus right when it's most needed, isn't it?

In the attached log, modem disconnects at 08.58.19 timestamp
[log202304261031.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/11361378/log202304261031.log)

broukema, 2023-09-07:
@tiol11 I've posted https://github.com/the-modem-distro/pinephone_modem_sdk/issues/213 . The USB disconnection lines in relation to the incoming call lines look similar to my case.

Your overall system (kde/plasma instead of phosh; hfd-service instead of feedbackd) is different to mine.  The line where I see the logout request starting in your log is
````
apr 26 08:58:53 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
apr 26 08:58:53 mobian dbus-daemon[1158]: [session uid=1000 pid=1158] Activating service name='org.kde.LogoutPrompt' requested by ':1.24' (uid=1000 pid=1306 comm="/usr/bin/ksmserver")
apr 26 08:58:54 mobian dbus-daemon[1158]: [session uid=1000 pid=1158] Successfully activated service 'org.kde.LogoutPrompt'
apr 26 08:58:55 mobian org.kde.LogoutPrompt[2302]: qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
apr 26 08:58:55 mobian org.kde.LogoutPrompt[2302]: qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
apr 26 08:58:57 mobian dbus-daemon[1158]: [session uid=1000 pid=1158] Activating service name='org.kde.Shutdown' requested by ':1.75' (uid=1000 pid=2302 comm="/usr/lib/aarch64-linux-gnu/libexec/ksmserver-logou")
apr 26 08:58:57 mobian dbus-daemon[1158]: [session uid=1000 pid=1158] Successfully activated service 'org.kde.Shutdown'
````
`ksmserver` appears to be the plasma session manager: https://develop.kde.org/docs/features/session-managment

So the session manager says that it gave a logout prompt and a few seconds later started a shutdown, just after `hfd-service` gave you a rumble (vibration). Just a bit earlier there is
````
apr 26 08:58:46 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
apr 26 08:58:46 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
apr 26 08:58:46 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
apr 26 08:58:46 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
apr 26 08:58:46 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
apr 26 08:58:47 mobian hfd-service[414]: rumbling with magnitude: 24576 for 100ms
````

So a guess for the shutdown is that `hdf-service` was still giving you vibrations long after the USB had been disconnected at 08:58:23, you tapped the screen to try to stop the vibrations, and that was interpreted by `ksmserver` as agreeing to logout and shut down. Another wild guess is that `ksmserver` found too many problems to handle and decided to logout + shutdown after giving the user just a few seconds to try to say no.

I'm on Mobian/trixie with `phosh` and `feedbackd`; the part in common that I see is the pattern of USB disconnections on an incoming call.

tiol11, 2023-09-10:
@broukema thank you for analyzing my log!

After that lost call, I had to give up for some time and get back to Android: my wedding was less than 1 month ahead and I couldn't afford to lose any further call... Later I got back to OG PP with Mobian bookworm and Phosh environment. One of major problems with Plasma mobile being bit heavier than Phosh, that can be felt on a resource-limited device like the OG PP: accepting incoming calls was a painful experience, especially given a 2-3 seconds delay between the ringing notification starting and the "accept call" button appearing. Not giving up with Plasma mobile though (I am also maintaining those packages in Debian!), just delaying it to when I will be able to daily-drive a more powerful phone...

Back to the topic, I haven't experienced this bug any more on my device. Likely by chance, as I do not take/make that much phone calls daily. Maybe you @broukema make much more calls and you have higher chance to hit this issue, also higher chance to try and debug it...

Closing this as a duplicate of #213

closed 2023-09-10 by tiol11

broukema, 2023-09-10:
@tiol11 Congratulations for your marriage! :) Glad to hear you're maintaining some plasma mobile packages in Debian. I don't make that many calls, but Biktorgj wants to chase up this bug, so maybe we'll get there. :)