the-modem-distro/pinephone_modem_sdk #210: After modem randomly goes dead, 30% of the time and won't recover

ell1e, 2023-07-29:
After a while on my Allwinner 3GB PinePhone with postmarketOS 23.06, the modem will disappear and then have a chance of staying dead of maybe roughly 30%, and when it does it'll just be gone forever from the notifications bar and gnome control center, and never come back even with hours of waiting. This is a definite downside and worsening compared to Quectel stock, where for me the modem will similarly die e.g. after suspend multiple times a day but only very rarely (maybe once every 1-2 weeks) ever not come back a minute or two later.

[2023-07-39-dmesg-oss-modem-resume.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12207203/2023-07-39-dmesg-oss-modem-resume.txt)

Alaraajavamma, 2023-08-24:
I did have the same problem with PP/pmos/Phosh/edge but I tried to enable all recommended udev settings and it did the trick. Before the changes I did get maybe 2-6 hours reliable modem and after changes I have now three days without zero issues (zero dropdowns and I get all calls and sms because I can monitor them trough my carrier network interface also)  

ACTION=="add", SUBSYSTEM=="usb", DRIVERS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/control}="on"
ACTION=="add", SUBSYSTEM=="usb", DRIVERS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{power/persist}="1"
ATTRS{idVendor}=="2c7c", ATTRS{idProduct}=="0125", ATTR{avoid_reset_quirk}="0"

ell1e, 2023-09-13:
Is this a postmarketOS packaging bug then? Should those rules be used by everyone? I wonder what the "persist" does, wouldn't that effectively disable the modem from suspending, is that actually a good thing to do?