the-modem-distro/pinephone_modem_sdk #216: Modem Not Working

Peter2469, 2023-09-12:
It seems that for both phosh and plasma mobile, I cannot connect to the modem and use it.
It is shown in lsusb and the general settings however that it does exist which means it must not be communicating fully.

I have provided logs and I would appreciate the help to getting my modem fully operational again

[dmesg-modem.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12585103/dmesg-modem.txt)
[dmesg-pinephone.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12585104/dmesg-pinephone.txt)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12585105/eg25-manager.log)
[modemmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12585106/modemmanager.log)
[networkmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12585107/networkmanager.log)
[openqti.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12585108/openqti.log)

Biktorgj, 2023-09-12:
It seems to be just looking for the network but not finding a valid network to connect to:
```
Sep 12 10:49:32 pinephone ModemManager[3597]: <info>  [modem1] 3GPP registration state changed (unknown -> unknown)
Sep 12 10:49:33 pinephone ModemManager[3597]: <info>  [modem1] state changed (enabling -> enabled)
Sep 12 10:50:36 pinephone ModemManager[3597]: <info>  [modem1] simple connect started...
Sep 12 10:50:36 pinephone ModemManager[3597]: <info>  [modem1] simple connect state (6/10): register
Sep 12 10:50:36 pinephone ModemManager[3597]: <info>  [modem1] 3GPP registration state changed (unknown -> searching)
Sep 12 10:50:36 pinephone ModemManager[3597]: <info>  [modem1] state changed (enabled -> searching)
Sep 12 10:51:37 pinephone ModemManager[3597]: <info>  [modem1] 3GPP registration state changed (searching -> idle)
Sep 12 10:51:37 pinephone ModemManager[3597]: <info>  [modem1] state changed (searching -> enabled)
Sep 12 10:52:17 pinephone ModemManager[3597]: <info>  [modem1] 3GPP registration state changed (idle -> searching)
Sep 12 10:52:17 pinephone ModemManager[3597]: <info>  [modem1] state changed (enabled -> searching)
```
Have you opened the phone recently (so maybe a wire may be disconnected on the main board?)
If you put the SIM in another phone does it work?

Peter2469, 2023-09-12:
I have opened the phone recently but this issue has been happening even before I opened it up (There was no disconnected wires); I put the sim into the other phone and it works fine. Both sim cards don't get recognised properly in the PP but instantly detected fine on my Pixel

Peter2469, 2023-09-12:
Surprisingly on rare occasions, the modem does actually connect successfully and the modem settings is all correct however I was unable to send a text/call/use data which means that something is broken still.
I restarted to Phosh to see if it was working there but it wasn't and ofc restarting back to plasma now means that the modem is still struggling to even connect.

LukeFilewalker78, 2023-09-12:
Hi, I had the same problem last week, I thought it was my Sim Card, it has like 8 years old, so I requested a new one, before that, I tried many times my old one with Mobian Bookworm 12, which it was the one I was using at that moment and Manjaro Phosh with no result, I tried other Sim that actually it worked so I decided to renew the Sim Card, but today it wouldn't work after activation, so I switch off the modem, and then I restarted eg25-manager via systemctl on the terminal, and then switched on the modem again and it's working again, it was pretty weird since I've been using the distro for a month with no problem and the community firmware 0.7.4 and band base 30.006.30.006, I already see a new band version, in the last release page, I hope it helps.

Peter2469, 2023-09-12:
Flashed [30.007.30.007](https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_30.007.30.007/update/NON-HLOS.ubi) and "Warning: skip copying modem image avb footer (modem partition size: 0, modem image size: 41025536)." appears; I know this exists when I updated the firmware also but I don't think that is the main issue.

@LukeFilewalker78 I followed your process and the modem didn't start so I restarted the device and it unfortunately didn't work for me but thanks anyway

LukeFilewalker78, 2023-09-12:
Thanks for the warning, I will wait then to update it, I saw many reports in the last days about the modem disconnecting, so I feared it was something more complicated, in my case there wasn't a carrier or a network type, I'd tried changing the network mode to 2G,3G, flashed the firmware again and my next step it was going to be downgrading to stock version, but it did work to my this way, sorry not be of more help.

LukeFilewalker78, 2023-09-12:
In that case it will be better to wait a bit, maybe the issue would be addressed soon, to be honest the only distro that work really good in my experience is Mobian, I did open my phone as well and I didn't have any problem hopefully.

Peter2469, 2023-09-12:
After keeping my phone idle and charged these are the logs; The Modem still does not connect

[dmesg-modem.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12589362/dmesg-modem.txt)
[dmesg-pinephone.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12589363/dmesg-pinephone.txt)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12589365/eg25-manager.log)
[modemmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12589366/modemmanager.log)
[networkmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12589367/networkmanager.log)
[openqti.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12589368/openqti.log)

Peter2469, 2023-09-19:
It does seem like it is stuck at searching and getting timed out however if I ask it select manually instead of automatically on very rare occasion O2 does show up which is correct for this instance but clicking on it does nothing and it timed out regardless.

Peter2469, 2023-09-19:
This was the response when I did flashall

[flashall_response.txt](https://github.com/the-modem-distro/pinephone_modem_sdk/files/12662731/flashall_response.txt)