the-modem-distro/pinephone_modem_sdk #222: All people I ever talk to complain I'm very quiet (jack plug, headset, pulseaudio/postmarketOS)

ell1e, 2023-10-20:
All the people I ever talk to complain I'm very quiet, which I think I've seen less with the stock firmware. I use a headset via the  jack audio plug which I know records fine since I checked on a laptop via Audacity, also I checked during multiple calls that the gnome-control-center sound recording settings have the mic boost at literally max volume, so it must be an error in the software stack past that. This affects 2G calls since 3G doesn't exist and due to emergency calls not working via 4G here and the fallback for not-4G-calls not working at all, for safety reasons only 2G calls can be used in Germany safely (disclaimer: at least according to my knowledge, I'm no expert on the matter don't trust me this isn't like, safety advice). It happens on postmarketOS which  I think as of now still uses pulseaudio to route audio.

mikeshenson, 2023-12-01:
What phone do you have? 

On my PinePhone Pro I made these changes to increase the volume.

sudo nano /usr/share/alsa/ucm2/PinePhonePro/VoiceCall.conf
#Set
cset "name='ADC Capture Volume' 160"
...
cset "name='IN1 Boost' 0"
cset "name='IN2 Boost' 1"
cset "name='IN3 Boost' 0"

ell1e, 2023-12-01:
Sorry, I forgot to specify! I have the 3GB Allwinner modem.

So is this a distribution bug that this isn't set by default? I'm using an external headset however, so if this affects the internal headset mic then it won't help with anything.

ell1e, 2023-12-01:
Interesting! Sadly I do, with the external headset and this non-standard firmware. (I don't use the internal one so I wouldn't know if that's affected to.)

ell1e, 2026-01-03:
This has been fixed, I think. Or at least whenever it still occurs, I think it's more of an issue of reception than anything else.