the-modem-distro/pinephone_modem_sdk #223: PPP: Call audio heavily distorts

UltraBlackLinux, 2023-10-27:
Hey there,
I currently cannot use my PPP for calling, because the call audio heavily distorts, seemingly on both sides: Call audio seems to muffle and reduce in volume to a point where nobody can understand anyone. Sometimes it still manages to recover for maybe three seconds, just before going back to being heavily muffled.

Any idea what's going on? I'm running version 0.7.4

mikeshenson, 2023-12-01:
Distorts sounds like you have it on speaker phone. Do not use the speaker phone on the PinePhone Pro.

On my PinePhone Pro I made these changes to increase the volume.

sudo nano /usr/share/alsa/ucm2/PinePhonePro/VoiceCall.conf
#Set
cset "name='ADC Capture Volume' 160"
...
cset "name='IN1 Boost' 0"
cset "name='IN2 Boost' 1"
cset "name='IN3 Boost' 0"

UltraBlackLinux, 2023-12-01:
I don't have the speaker enabled, it really is on the earpiece