the-modem-distro/pinephone_modem_sdk #228: Move code to GitLab or Codeberg

julianfairfax, 2024-01-02:
It would be nice if the project's code were moved to GitLab or Codeberg. GitHub is not an open source platform, and being owned by Microsoft makes it more of a danger to open source projects, rather than a help.

broukema, 2024-03-02:
Fully agree. More details at https://giveupgithub.org . I seem to remember a chat with @Biktorgj saying that he supports this in principle, but he hasn't been able to get around to it. These can be purchased at, for example, https://www.21stkeys.co.nz/Keys/Round%20Toit.htm and sent to all friends and colleagues (and yourself) who don't have round toits (yet).

giorez, 2024-04-25:
totally agree to move away. since the project seems to stall, any help to make it happen?