the-modem-distro/pinephone_modem_sdk #238: Pinephone Pro V02 2023-0909 - Firmware upgrade failed

HodlOnToYourButts, 2024-09-14:
I attempted to upgrade the modem firmware on my new Pinephone Pro using fwupdmgr, but now it's refusing to show up at all. I opened the back hoping to find and short the EDL pins, but was confronted with a metal cage over where they were shown in the picture on the wiki. It came off relatively easily, but to my horror there are no pins to short. Did I just brink my modem? Please help.

HodlOnToYourButts, 2024-09-14:
[image]

Zapeth, 2024-09-25:
The picture on the wiki shows the pins for the regular Pinephone, it seems they are positioned elsewhere on the Pinephone Pro -> https://forum.pine64.org/showthread.php?tid=19007