the-modem-distro/pinephone_modem_sdk #246: phone audio broken on reboots with v8.0 firmware

henrythemouse, 2025-01-12:
Hi,

A  few weeks back I installed v8.0 and found that after rebooting my PPP I had no audio on phone calls (including messages).  At first I would just re-run the install script after the reboot and then the phone would have audio again. After that I discovered by accident that if I powered off the PPP and powered it back on, the phone audio was fine. 

This is where I'm at now, I always power off the phone, because if I reboot it the phone has no audio. Of course, sometimes I forget ;-). I'm posting this to see if anyone else has seen this happen on a PPP.  If so I'd like to know if they might have a solution. 

 This is an updated PPP running Mobian on Debian trixie.

blackmagic0, 2025-01-13:
I am having a similar issue on PPP/mobian but I have not had any working audio for phone calls since installing v0.8.0 at all.  I tried adjusting audio settings in `/usr/share/alsa/ucm2/PinePhonePro/VoiceCall.conf` to increase volume on the built in mic but it didnt work.  restarting or power off holding power button did not fix audio either.

blackmagic0, 2025-01-14:
> Nice to have confirmation that this issue is repeatable. Thanks for the write up. The path on my phone to VoiceCall.conf is slightly different (a typo perhaps).
> 
> `/usr/share/alsa/ucm2/Rockchip/PinePhonePro/VoiceCall.conf`
> 
> Maybe I have altered this file and forgotten what I did or maybe my VoiceCall.conf is the original. In any case I'm attaching the file here for reference. When reading this file I noticed some settings that were outside of the recommended settings (per the comments). When using the phone I don't have any problems with volume or mic that I know of (maybe the other end of the call is having problems).
> 
> [VoiceCall.conf.txt](https://github.com/user-attachments/files/18412958/VoiceCall.conf.txt)

Mine is in the path I wrote, there is also a rockchip folder but it doesn't seem to have anything relevant.  I am on mobian-pinephonepro-phosh-12.0 downloaded on 2025-01-10

its also a link or whatever to /usr/share/alsa/ucm2/Pine64/PinePhonePro/

[image]
[image]
[VoiceCall.conf.txt](https://github.com/user-attachments/files/18413473/VoiceCall.conf.txt)

henrythemouse, 2025-01-31:
This is strange, maybe it's Trixie problem. 

You did say that you did a new install of mobian and my installation is likely 8-10 months old. Still even with this path difference, we share the same poweroff / reboot  issue, when it comes to this firmware version. I have several intermittent  problems with my PPP, but none that depend on poweroff. 

 I don't have a PindPhonePro folder in the ucm2 folder, but I do in the Rockchip folder ( which has the same files that you have under ucm2 / PinePhonePro ).

````
/usr/share/alsa/ucm2$ ls

Allwinner  common       Intel     OMAP       Rockchip       Tegra
AMD        conf.d       lib       OnePlus    Samsung        ucm.conf
Amlogic    conf.virt.d  MediaTek  platforms  sdm845         USB-Audio
blobs      DEBUG.md     module    Qualcomm   SHIFT          Xiaomi
codecs     HDA          NXP       README.md  sof-soundwire

````

````
/usr/share/alsa/ucm2/Rockchip/PinePhonePro$ ls

HiFi.conf  PinePhonePro.conf  VoiceCall.conf

````

I've just updated my mobian installation, and after checking, it still has to be powered off / on in order to make or receive calls with audio. The phone app does seem to connect and time ticks away, but no audio. I'm thinking that this may not be a firmware problem, more a problem with Trixie / Mobian.

I'll let this stay open for a while in case anyone else has some ideas.

mpbagot, 2025-02-03:
Just as another data point, I am running Danctnix Arch Linux ARM on a late 2024 Pinephone Pro. I've got all the latest updates from the repos.
I have this exact same problem, and have since I installed the initial image. Rebooting doesn't fix the audio, but shutting down and powering on again does.
My VoiceCall.conf is almost identical aside from the ADC Capture Volume and the IN1 Boost., which are 47 and 8 respectively.

henrythemouse, 2025-02-05:
As of update on 2025-02-03 this work around does not apply. The sound configuration on the PPP has changed and along with that the phone has no sound at all, including when cycled with Power Off.

henrythemouse, 2025-02-08:
I can't see that the modem firmware is at fault here. Closing.

closed 2025-02-08 by henrythemouse

henrythemouse, 2025-02-18:
Just in case someone needs an option,  I installed Phosh ARM/DanctNIX  on my PPP and had a working phone/sms instantly. No camera however.

[Arch Phosh](https://github.com/dreemurrs-embedded/Pine64-Arch)